skycoin/hardware-wallet #2: [Hardware Wallet]: USB protocol's vendor Id, product Id and manufacturer name are wrong

mpsido, 2018-08-27:
The skycoin firmware is still using "Satoshi Lab" 's identifiers in the usb protocol.
Skycoin product need to decide their own

mpsido, 2018-08-27:
Brandon: Change to "Skycoin Foundation" from satoshi labs

mpsido, 2018-10-03:
Skycoin Foundation is not listed in the usb vendors here:
http://www.linux-usb.org/usb.ids

mpsido, 2018-10-03:
> Skycoin Foundation is not listed in the usb vendors here:
> http://www.linux-usb.org/usb.ids

I dont really know the process to get an official vendorId but a quick google search led me here:
https://www.usb.org/getting-vendor-id

mpsido, 2018-11-26:
Despite changing the udev rules as following:

```
## 0483:df11 STMicroelectronics STM Device in DFU Mode
SUBSYSTEM=="usb", ATTR{idVendor}=="0483", ATTR{idProduct}=="df11", MODE="0666"
## 0483:3748 STMicroelectronics ST-LINK/V2
SUBSYSTEM=="usb", ATTR{idVendor}=="0483", ATTR{idProduct}=="3748", MODE="0666"
## 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)
SUBSYSTEM=="usb", ATTR{idVendor}=="0483", ATTR{idProduct}=="374b", MODE="0666"
## 534c:0001 SkycoinFoundation Wallet [SKYCOIN]
SUBSYSTEM=="usb", ATTR{idVendor}=="534c", ATTR{idProduct}=="0001", MODE="0666"
KERNEL=="hidraw*", ATTRS{idVendor}=="534c", ATTRS{idProduct}=="0001", MODE="0666"
## 1209:53c0 SkycoinFoundation SKYCOIN v2 Bootloader
SUBSYSTEM=="usb", ATTR{idVendor}=="1209", ATTR{idProduct}=="53c0", MODE="0666"
KERNEL=="hidraw*", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="53c0", MODE="0666"
## 1209:53c1 SkycoinFoundation SKYCOIN v2
SUBSYSTEM=="usb", ATTR{idVendor}=="1209", ATTR{idProduct}=="53c1", MODE="0666"
KERNEL=="hidraw*", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="53c1", MODE="0666"
```

The command 
```
lsusb -d 534c:0001
``` 
Still returns this:
```
Bus 001 Device 104: ID 534c:0001 SatoshiLabs Bitcoin Wallet [TREZOR]
```

Which confirms the vendor id is not declared only by the device, but at the USB protocol level

mpsido, 2018-11-26:
```
lsusb -v -d 534c:0001
```

```
Bus 001 Device 104: ID 534c:0001 SatoshiLabs Bitcoin Wallet [TREZOR]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x534c SatoshiLabs
  idProduct          0x0001 Bitcoin Wallet [TREZOR]
  bcdDevice            1.00
  iManufacturer           1 SkycoinFoundation
  iProduct                2 SKYCOIN
  iSerial                 3 AB1D4DE63F5E6CB978273207
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              4 SKYCOIN Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

mpsido, 2018-11-26:
[image]
The idVendor and idProduct come from protocol, iManufacturer iProduct are given by the device

mpsido, 2018-11-26:
Linux contains files that device the "official" vendor ids

```
grep Satoshi `locate usb.ids`
```
```
/usr/share/misc/usb.ids:534c  SatoshiLabs
/var/lib/usbutils/usb.ids:534c  SatoshiLabs
```
```
grep Satoshi `locate usb.ids`
```
```
/usr/share/misc/usb.ids:534c  SatoshiLabs
/var/lib/usbutils/usb.ids:534c  SatoshiLabs
```

mpsido, 2018-11-27:
If we don't have budget to get an official vendor ID we can use a non used plage. 
I could find one between SatoshiLabs and Meyer Instruments for instance:

```
534c  SatoshiLabs
	0001  Bitcoin Wallet [TREZOR]
5354  Meyer Instruments (MIS)
```

Whe can use one of these 534d 534e 534f 5350 5351 5352 5353 ....
That won't get a pretty "Skycoin Foundation" appearing on the OS's dialog box. But at least we won't be appearing as SatoshiLabs.

Dawn-Piper, 2018-12-06:
Sent vendor ID to @mpsido today

mpsido, 2018-12-06:
Received vendor Id:
```
The vendor ID number for Skycoin is:
Decimal: 12602
Hex: 313A
```

closed 2018-12-07 by mpsido