skycoin/hardware-wallet #9: [Hardware Wallet]: Break in trezor lock

mpsido, 2018-09-27:
Context: if we flash an official trezor firmware on the device, the first thing it does is lock the Memory Process Unit so that there is no way to change the bootloader.

The bootloader is the way they use to protect against attempt to install a fraudulent firmware on a trezor.
 All it does is checking firmware signature.

The firmware can only be updated using a dedicated message that change the firmware code without touching the bootloader.

For skycoin developement this security lock is a problem: we need to recover the devices that are "victim" of this protection mechanism so we can modify the bootloader and firmware freely according to our needs.

mpsido, 2018-09-27:
The MPU(Memory Process Unit) configuration that is locking the device is in mpu_config function (setup.c).

The idea of the hack is to create a firmware that implement a message able to execute another configuration of the MPU.

1. Make sure it is possible to update the firmware using the normal method. Try to write in a "true" trezor firmware that was locally compiled.
2. Modify that firmware creating a new message that can be executed from our golang CLI.
3. write our version of the MPU configuration and execute at message reception

mpsido, 2018-09-27:
1. To update the firmware we need to switch to "bootloader mode"

mpsido, 2018-09-27:
The command line to upload firmware is :
    ./trezorctl firmware_update -f fw.bin

See here https://github.com/trezor/python-trezor to install trezorctl

mpsido, 2018-09-27:
2. this file hacks trezorctl to add a new message mpu_cheat  in the trezorctl CLI

mpsido, 2018-09-27:
2. this file is the patch on firmware side that makes it able to listen to the request we added on trezorctl CLI
[mpucheat.patch.txt](https://github.com/skycoin/hardware-wallet/files/2422805/mpucheat.patch.txt)

mpsido, 2018-09-27:
3. found some documentation [here](http://comm.eefocus.com/media/download/index/id-1008313)

mpsido, 2018-09-27:
It is not guaranteed that that trick can succeed because the configuration done by the bootloader is leaving priviledged mode.

But even if it doesn't work we can at least program it for skycoin. But BIG WARNING **don't do that until the firmware update function is ported back in skycoin firmware** otherwise we shut our only open door for update.

mpsido, 2018-09-27:
3. inserting this configuration function call in the processing of "mpu cheat message"

    void mpu_unlock(void)
    {
    #if 1
        // Disable MPU
        MPU_CTRL = 0;

        // Note: later entries overwrite previous ones
        // Flash (0x08000000 - 0x0807FFFF, 1 MiB, read-only)
        MPU_RBAR = FLASH_BASE | MPU_RBAR_VALID | (0 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_FLASH | MPU_RASR_SIZE_1MB | MPU_RASR_ATTR_AP_PRW_URW;

        // Metadata in Flash is read-write when unlocked
        // (0x08008000 - 0x0800FFFF, 32 KiB, read-write)
        MPU_RBAR = (FLASH_BASE + 0x8000) | MPU_RBAR_VALID | (1 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_FLASH | MPU_RASR_SIZE_32KB | MPU_RASR_ATTR_AP_PRW_URW;

        // SRAM (0x20000000 - 0x2001FFFF, read-write)
        MPU_RBAR = SRAM_BASE | MPU_RBAR_VALID | (2 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_SRAM | MPU_RASR_SIZE_128KB | MPU_RASR_ATTR_AP_PRW_URW;

        // Peripherals (0x40000000 - 0x4001FFFF, read-write)
        MPU_RBAR = PERIPH_BASE | MPU_RBAR_VALID | (3 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_PERIPH | MPU_RASR_SIZE_128KB | MPU_RASR_ATTR_AP_PRW_URW;
        // Peripherals (0x40020000 - 0x40023FFF, read-write)
        MPU_RBAR = 0x40020000 | MPU_RBAR_VALID | (4 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_PERIPH | MPU_RASR_SIZE_16KB | MPU_RASR_ATTR_AP_PRW_URW;
        // Flash controller is protected
        // (0x40023C00 - 0x40023FFF, privileged read-write)
        MPU_RBAR = 0x40023c00 | MPU_RBAR_VALID | (5 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_PERIPH | MPU_RASR_SIZE_1KB | MPU_RASR_ATTR_AP_PRW_URW;
        // Don't enable DMA controller access
        // Peripherals (0x50000000 - 0x5007ffff, read-write)
        MPU_RBAR = 0x50000000 | MPU_RBAR_VALID | (6 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_PERIPH | MPU_RASR_SIZE_512KB | MPU_RASR_ATTR_AP_PRW_URW;
        // SYSCFG_* registers are disabled
        // (0x40013800 - 0x40013BFF, read-write)
        MPU_RBAR = 0x40013800 | MPU_RBAR_VALID | (7 << MPU_RBAR_REGION_LSB);
        MPU_RASR = MPU_RASR_ENABLE | MPU_RASR_ATTR_PERIPH | MPU_RASR_SIZE_1KB | MPU_RASR_ATTR_AP_PRW_URW;

        // Enable MPU
        MPU_CTRL = MPU_CTRL_ENABLE | MPU_CTRL_HFNMIENA;

        // Enable memory fault handler
        SCB_SHCSR |= SCB_SHCSR_MEMFAULTENA;

        __asm__ volatile("dsb");
        __asm__ volatile("isb");

        // Switch to unprivileged software execution to prevent access to MPU
        // set_mode_unprivileged();
    #endif
    }

it is executed DOESN'T FIX our issue

mpsido, 2018-09-27:
Did not get better after adding a call to svc_flash_unlock before calling mpu_unlock:

`
void fsm_msgMPUCheat(const MPUCheat* msg)
{
	(void)(msg);
	svc_flash_unlock();
	mpu_unlock();
	fsm_sendSuccess("Message received");
}
`

mpsido, 2018-09-27:
Trying to move the configuration function inside svc call to get higher priviledges. See this patch: 
[mpucheat_svc.patch.txt](https://github.com/skycoin/hardware-wallet/files/2423852/mpucheat_svc.patch.txt)

mpsido, 2018-09-28:
> It is not guaranteed that that trick can succeed because the configuration done by the bootloader is leaving priviledged mode.
> 
> But even if it doesn't work. We can at least use firmware update message to program the device for skycoin. But BIG WARNING **don't do that until the firmware update function is ported back in skycoin firmware** otherwise we shut our only open door for firmware update.
> 
> Related to issue #8

Just checked, the firmware update and firmware erase messages are inside bootloader code, the firmware can be updated safely

mpsido, 2018-09-28:
In trezor the firmware also checks bootloader validity: bl_check.c function check_bootloader. It checks using hash fingerprint

mpsido, 2018-09-28:
in bl_check.c there is also some code that is able to update bootloader...

mpsido, 2018-11-26:
memory.h contains a memory map, maybe we can find a hack in firmware update allowing to write code at a different address

mpsido, 2018-12-17:
Consider following the steps of the tutorial over there: https://www.youtube.com/watch?v=ZUXuk2zFHfs

mpsido, 2018-12-17:
This video provides another way to explain how to use the system bootloader : https://www.youtube.com/watch?v=boXSdDnHaAE

jdknives, 2019-02-25:
Is this issue still valid? It seems like we can manipulate the bootloader already, so this issue seems outdated. Or am I misunderstanding the concern that is stated in the initial comment?

closed 2019-09-20 by jdknives