skycoin/hardware-wallet #14: Device not recognised by st-link

Dawn-Piper, 2018-10-24:
St-link is recognized by computer, devices are not. When flashing previously I wrote a guide, I tried the guide on a fresh machine to confirm that it worked, Senyoret then followed that guide, also with success. Now, it does not work.

I've tried:
1. Multiple devices, including ones that have been successfully flashed in the past and also fresh devices from that batch.
2. Multiple st-link programmers
3  Multiple machines
4. Windows and linux
5. With and without the RST pin connected
6. With and without usb cable connected

st-info --probe output: Found 1 stlink programmers

mpsido, 2018-10-24:
quote from @Dawn-Piper : the firmware were flashed with skycoin firmware and the "wrong firmware signature" warning do not appear on these devices

Dawn-Piper, 2018-10-25:
st-util output: 
2018-10-24T10:52:19 INFO common.c: Loading device parameters....
2018-10-24T10:52:19 WARN common.c: unknown chip id! 0x5fa0004

Dawn-Piper, 2018-10-26:
The pins on the C2 wallets are reverse of the B2 accounting for the devices not being recognised. There is no further issue here.

closed 2018-10-26 by Dawn-Piper