skycoin/hardware-wallet #17: [Hardware Wallet]: Incorrect logo at startup

Senyoret1, 2018-11-13:
Just after connecting the hardware wallet via USB, the Trezor logo is displayed for a brief moment and then is replaced by the Skycoin logo. That could cause problems in the future.

mpsido, 2018-11-17:
need to change the logo in the bootloader

mpsido, 2019-01-25:
quote from @Senyoret1
Today I updated the wallet using the bootloader and saw that when turning on the wallet with booth buttons pressed the Trezor logo is displayed. Maybe it should be removed

Also, after updating it says that I can unplug the "Skycoin W"

mpsido, 2019-01-25:
I removed the skycoin logo from the bootloader because the smallest bitmap file i found for it is still too big for the bootloader memory. even when removing every other bitmap.

The text message csn be fixed. and about the logo I suggest "no logo" then.

Senyoret1, 2019-01-25:
No logo may be good, as having the logo of another company may be problematic, but that is just my opinion. I can not make the decision.

olemis, 2019-01-25:
What size is ok ?

stdevAlDen, 2019-02-01:
> What size is ok ?

Let me check, with a data sheet and a hexdump could be possible determine the available size.

olemis, 2019-03-19:
@stdevStark cpould you pls get this done asap ?

stdevStark, 2019-03-19:
I have not device . I use the emulator . For this I need a device .

olemis, 2019-03-19:
@stdevStark right ...

olemis, 2019-03-21:
Fixed in 9f6aa3a

PR #137 (2019-03-21): fixes #17 - Skycoin branding graphics for SkyWallet devices
Fixes #17 

Changes:
- Replace former PNG logo bitmaps with Skycoin's
- Regenerate bitmap data 
- Do not fail when input PNG are 8-bit B/W images

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes

Tested manually in device.
commit 9f6aa3a7 (2019-03-21): Merge pull request #137 from simelo/olemis_t17_boot_skylogo_v2

fixes #17 - Skycoin branding graphics for SkyWallet devices
closed 2019-03-21 by olemis