skycoin/hardware-wallet #26: [Hardware-Wallet]: Support for translations

mpsido, 2018-11-30:
The firmware code contains tags on the strings intended to be displayed to user, that may allow translation in the future. 
Shall we work on this ?

Dawn-Piper, 2018-12-04:
not now

mpsido, 2018-12-21:
Check what we can do with this: https://github.com/zst-embedded/STM32-LCD_Font_Generator

mpsido, 2018-12-21:
Every text meant to be shown to the user in the code should be wrapped with `_()` , it ignored by C compiler thanks to gettext.h

mpsido, 2018-12-21:
I suspect the transformation of text goes with the scripts in [gen/fonts](https://github.com/trezor/trezor-mcu/tree/master/gen/fonts)

mpsido, 2018-12-25:
fonts.c include font.inc and fontfixed.inc
These two files contain the same letters but displayed a bit differently (which means we have two possible fonts)
fonts.c defines two functions fontCharWidth and fontCharData

These functions are called in : oled.c::oledDrawChar

We have two types of font: FONT_STANDARD and FONT_FIXED (defined in fonts.h)

In trezor-mcu/gen/fonts/ there is a script generate.py which transforms a png containing letters in the right order into fonts.inc files

mpsido, 2018-12-25:
Chinese characters map here: http://www.khngai.com/chinese/charmap/tbluni.php?page=5

mpsido, 2019-01-08:
I think I need like 2-3 more hours to code a map like structure plus the code that fetch the character representation in that map if we are using chinese fonts.

mpsido, 2019-01-10:
Created a pull request  #41  that " displays a "你好世界" message on the screen when the "GetVersion" message is received"

mpsido, 2019-01-11:
To get chinese translation I need :
* the exhausive list of all characters you are going to be using
* the font you wanna use (in a ttc of ttf file)