skycoin/hardware-wallet #27: [Hardware-Wallet]: cleaning deprecated trezor code

mpsido, 2018-11-30:
The code that need attention is:
* storage structure
* Features message
* u2f releated

mpsido, 2018-11-30:
What to do with fsm_msgResetDevice ?

stdevAlDen, 2019-02-22:
> What to do with
> 
>     * fsm_msgResetDevice ?
> 
>     * fsm_msgLoadDevice ?

https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/reset.h
https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/reset.c
https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/protob/messages.proto#L355

stdevAlDen, 2019-02-22:
> Also:
> 
>     * entropy

We are using entropy from user host but some code related look deprecated/unused

stdevAlDen, 2019-02-22:
>     * Features

Features msg is used, consider some fields...

stdevAlDen, 2019-02-22:
consider this [code block](https://github.com/skycoin/hardware-wallet/blob/230a1d5b76fb210506adc257b619bdd6201e5135/tiny-firmware/firmware/layout2.c#L109-L149)