skycoin/hardware-wallet #28: [Hardware-Wallet]: programmatically press emulator's buttons

mpsido, 2018-11-30:
This feature could be very useful to create automatic tests.

The automatic tests could be inserted in travis build to increase robustness

mpsido, 2018-11-30:
important files to start hacking:
emulator/udp.c
emulator/buttons.c

mpsido, 2018-12-05:
Even if I do create this possibility some actions like input pincode or recovery seed need an user reading what is written on the screen

Dawn-Piper, 2018-12-06:
we will add this at a later stage

olemis, 2019-02-05:
@stdevHan pls try this https://stackoverflow.com/questions/35760351/simulate-keyboard-button-press-sdl-library . If not possible to do it from a remote process then implement a web server in emulator SDL app and listen for test commands . One such command should be pressing a button (given its GUI component ID).

olemis, 2019-02-21:
@stdevIvr what's the progress on this ? This is blocking other issues e.g. skycoin/skycoin#1783 , skycoin/hardware-wallet-js#59 (see [build failure](https://travis-ci.com/skycoin/hardware-wallet-js/builds/101709851) for skycoin/hardware-wallet-js#62) .

stdevIvr, 2019-02-21:
I'm trying to trigger the button event but I haven't yet. What I did was to add a middleware in the emulator in order to intercept specific messages to press the button. That works, but the button isn't really pressed yet. Still working on that.

olemis, 2019-03-10:
fixed by 9a86d9b

PR #108 (2019-03-10): Auto press emulator buttons
Fixes #28

Changes:
- Added support for pressing emulator's buttons programmatically.

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes
commit 9a86d9bb (2019-03-10): Merge pull request #108 from simelo/stdevIvr_t28_auto_press_emulator_buttons

Auto press emulator buttons fixes #28
closed 2019-03-10 by olemis