skycoin/hardware-wallet #30: [Hardware-Wallet]: Verify the address on the hardware wallet screen

mpsido, 2018-12-04:
Firmware side equivalent for : https://github.com/skycoin/hardware-wallet-js/issues/24

mpsido, 2018-12-11:
@Senyoret1 what shall we do if we use this message to get many addresses, shall we display them one by one with button confirmation ?

Senyoret1, 2018-12-11:
Seeing how the UI works in wallet.trezor.io, I think that what happens internally is more or less like this:

- When the software wallet needs an address, it is requested to the hardware wallet, which returns the complete address without showing it on the screen or requiring the user to confirm the operation. Possibly in that call the hw can return several addresses at once, just as it happens with the Skycoin firmware, but it seems that wallet.trezor.io only requests one.

- Once the complete address is obtained, the software wallet shows only a part of it, next to a button that allows the user to see the complete address. When the user presses the button, the complete address is displayed both on a software wallet popup and on the hw screen. At that moment the user must confirm in the hw that the address is the same.

Taking that into account, the function that lets the user confirm the address in the hw screen seems to be different from the function that allows to obtain the addresses and I think that it just receives the index of one address at a time.

mpsido, 2018-12-18:
Do you mean we should not be supporting requesting many addresses at once ? 

The reason I've done it like this it because at the time I was working with the skycoin desktop wallet, I've seen that during wallet creation the first thing it does is: parsing the skycoin blockchain to see if one of the first 100 addresses of the wallet is the recipient of a transaction somewhere.

That would have been very messy to work with it on a hardware wallet if we had to get addresses one by one this is why I chose this implementation.

I suggest an implementation like this:
If asking for N addresses the electronic wallet prompts :
 "Do you agree to generate N addresses ? Yes/No"

If asking for 1, we can do the same as trezor: display it on the screen and ask use confirmation

Senyoret1, 2018-12-18:
Actually I think the solution is not to modify what is already done and simply add a new function that allows the user to see an address in the hardware wallet and accept it if it is correct, something like `showAddress(addressIndex)`. More or less like this (the QR code button is not really necessary):
[image]

After the user presses the continue button, the software wallet takes care of the rest.

mpsido, 2018-12-19:
@Senyoret1 I added an option on devAddressGen function allowing you to activate that message display. I hope it fix this issue. If not just reopen it.

closed 2018-12-19 by mpsido