skycoin/hardware-wallet #33: [Hardware-Wallet]: Readapt Recovery seed for skycoin needs

mpsido, 2018-12-04:
Here is the currently existing RecoveryDevice structure:

```
message RecoveryDevice {
	optional uint32 word_count = 1;				// number of words in BIP-39 mnemonic
	optional bool passphrase_protection = 2;		// enable master node encryption using passphrase
	optional bool pin_protection = 3;			// enable PIN protection
	optional string language = 4 [default='english'];	// device language
	optional string label = 5;				// device label
	optional bool enforce_wordlist = 6;			// enforce BIP-39 wordlist during the process
	// 7 reserved for unused recovery method
	optional uint32 type = 8;				// supported recovery type (see RecoveryType)
	optional bool dry_run = 10;				// perform dry-run recovery workflow (for safe mnemonic validation)
}
```

We won't use the following elements: `dry_run` `type`, `enforce_wordlist` 

We don't use these yet:
* `word_count`: (fixed to value 12) 
* `language`: #26 
* `label`: #32

An extra attention need to be given to : `passphrase_protection` `pin_protection` (shall we unable configuration of these things from this message)

mpsido, 2018-12-07:
Related to #27

mpsido, 2018-12-18:
Related to https://github.com/skycoin/hardware-wallet-js/issues/31

olemis, 2019-01-30:
in the desktop wallet 12 or 24
[image]

olemis, 2019-01-31:
@stdevAlDen in Trezor code 12, 18, 24
https://github.com/skycoin/hardware-wallet/blob/master/skycoin-api/trezor-tools/bip39.c#L142

olemis, 2019-02-13:
closed by 801806e

olemis, 2019-02-18:
@stdevAlDen [new issues](https://github.com/skycoin/skycoin/issues/2165#issuecomment-463348983) found .

olemis, 2019-02-19:
[False alarm](https://github.com/skycoin/skycoin/issues/2165#issuecomment-464921548). Closing this issue once again .

commit 801806eb (2019-02-13): Merge pull request #70 from simelo/stdevAlDen_t33_reshape_recovery_seed_for_skycoin_needs

Reshape recovery seed for skycoin needs. fixes #33
closed 2019-02-19 by olemis