skycoin/hardware-wallet #34: [Hardware-Wallet]: Make a guide for firmware upgrade release

TienZhao, 2018-12-05:
We want to have a step-by-step guide for releasing firmware upgrade versions. The guide is supposed to start from listing all the resources needed, and ends at having a file that everybody can download for upgrade.

mpsido, 2018-12-11:
I am still wondering what you want in it. All the steps are described in the tiny-firmware/README.md file.
So I believe it is too complicated for non developer. But if so someone has to do the compilation anyway.

I was considering creating a docker file doing the compilation for you. but to use it you need to have docker installed and be able to run it.

mpsido, 2018-12-11:
Process to update firmware for users: https://wiki.trezor.io/index.php?title=User+manual%3AUpdating+the+Trezor+device+firmware&ModelType=1

mpsido, 2019-01-15:
Once issue #50 is solve this issue will be deprecated.
To create a release it will be just : `make release` command

closed 2019-01-15 by mpsido