skycoin/hardware-wallet #38: [Hardware-wallet] show transaction details on the screen

Senyoret1, 2018-12-21:
It is necessary to add to the hardware wallet a function to sign transactions and that function must show the details of the transaction on the hw screen and asks the user to confirm it. The function could receive the transaction data as JSON or serialized, whatever is more convenient.

Additionally, it would be good if the function for signing text strings also requests confirmation from the user on the hardware wallet, so that it can not be used to sign transactions without the user being adequately warned.

mpsido, 2019-01-14:
@Senyoret1 please as you are on it, can you give me "as much examples as you can" preferably both serialized and with the content in a JSON format. I would need around 10 so I can make automatic test out of it.
I can spend some time generating some on my station but I won't be sure it is the actual format you are using, and I think that is easier for you because you can just add a console.log instruction inside the function of my API that you are using: a `console.log(message);` instruction in `createSignMessageRequest` function line 369 device-wallet.js would be a very good start.

Thanks

Senyoret1, 2019-01-14:
I think that 10 examples won’t help you to create automatic tests, since the signatures, at least in theory, are not deterministic. Each time the result should be different.

Taking that into account, I will explanatin the current process for creating transactions and leave an example, maybe that is more useful. However, if you still need more sample txs, let me know and I'll create more.

The function that creates the transactions is this:
https://github.com/skycoin/skycoin/blob/5ae4a3e726f728ae33e3977ca72afe250b997d83/src/gui/static/src/app/services/wallet.service.ts#L295-L389

Most of the code is not important in this case, since what it does is to select the inputs and create the outputs. The first relevan part is line 365, which simply gets the inner hash of the transaction that is going to be created. The code of the function that returns the hash is in another repository, it is in here:
https://github.com/Senyoret1/skycoin-lite/blob/8d062afd37f9bf7554bf7264f14e1e6782742489/liteclient/client.go#L92-L96

What the code does is create the complete transaction, but without signing it, and then calculates the internal hash. The code uses Skycoin Cipher, so the procedure is the same used by the Skycoin node.

After getting the inner hash, line 368 of wallet.service.ts creates, recursively, an array with all the signatures nedded for the transaction. Each signature is created in here:
https://github.com/skycoin/skycoin/blob/5ae4a3e726f728ae33e3977ca72afe250b997d83/src/gui/static/src/app/services/wallet.service.ts#L505-L512

As you can see, what is signed is a SHA256 hash created with the inner hash and the hash of the input (so there is one signature for each input). The `AddSHA256` function is in Skycoin Cipher and is converted automaticaly to Javascript in the Skycoin-Lite repo.

After having all the signatures, the JS code simply calls this function to create the entire signed transaction:
https://github.com/Senyoret1/skycoin-lite/blob/8d062afd37f9bf7554bf7264f14e1e6782742489/liteclient/client.go#L78-L88

That function creates the transaction just like the function that returns the inner hash, but includes the signatures and returns the transaction encoded.

The functions used to create the transactions just receive the inputs and outputs of the transaction, so maybe that is what the hardware wallet should receive, but that could be changed if necesary.

A data example is this:

We could have this as input:
```
[{"hash":"99a1a50ffa21ab48ee7c31d01e7e14451f9834f5294468bd17e87c5018900b81","secret":"","address":"d1hMF1XCCvFXVa2u7NbuWo9dmfNbdpoFLJ","address_index":0,"calculated_hours":15,"coins":127}]
```

And this oputs:
```
[{"address":"d1hMF1XCCvFXVa2u7NbuWo9dmfNbdpoFLJ","coins":125000000,"hours":4},{"address":"2kVVoMkH7aTVXsiGwZEALpkHZ6sUyumL8hH","coins":2000000,"hours":3}]
```

The inner hash for the transaction would be this:
```
0dcbbaf454e158cc87330900080315b9f288b5ceb2d1047299fabc4c407c1a18
```

The final signed transaction would be:
```
{Length:220 Type:0 InnerHash:{Length:220 Type:0 InnerHash:[13 203 186 244 84 225 88 204 135 51 9 0 8 3 21 185 242 136 181 206 178 209 4 114 153 250 188 76 64 124 26 24] Sigs:[[77 66 96 193 40 92 239 6 100 51 95 212 23 34 102 201 127 208 9 114 68 45 45 168 136 92 174 182 209 89 38 195 95 27 84 49 74 167 100 10 71 189 217 66 198 111 226 97 84 69 174 23 37 56 93 121 225 55 188 202 222 220 161 182 1]] In:[[153 161 165 15 250 33 171 72 238 124 49 208 30 126 20 69 31 152 52 245 41 68 104 189 23 232 124 80 24 144 11 129]] Out:[{Address:d1hMF1XCCvFXVa2u7NbuWo9dmfNbdpoFLJ Coins:125000000 Hours:4} {Address:2kVVoMkH7aTVXsiGwZEALpkHZ6sUyumL8hH Coins:2000000 Hours:3}]}
```

The encoded version of that transaction is this:
```
dc000000000dcbbaf454e158cc87330900080315b9f288b5ceb2d1047299fabc4c407c1a18010000004d4260c1285cef0664335fd4172266c97fd00972442d2da8885caeb6d15926c35f1b54314aa7640a47bdd942c66fe2615445ae1725385d79e137bccadedca1b6010100000099a1a50ffa21ab48ee7c31d01e7e14451f9834f5294468bd17e87c5018900b810200000000597f682bf9d6302fc070eb0cee7c1c6a27653f214059730700000000040000000000000000fc3a66b52bb478be3a62bf8a698a64fa2ffbdedc80841e00000000000300000000000000
```

mpsido, 2019-01-18:
This [piece of code](https://github.com/skycoin/skycoin/blob/5ae4a3e726f728ae33e3977ca72afe250b997d83/src/gui/static/src/app/services/wallet.service.ts#L500-L508) shows that the "message" sent to the device to be signed is a "hash". Which means the device can't extra transaction parameters out of it.

I don't see how to make it work without having to change the way we talk to the device. 
I am gonna need to create a new function (or change the API of the existing signature function).
Instead of sending the hash of the transaction for the device to sign, we need to send the full json and expect the device to hash then sign.

Senyoret1, 2019-01-18:
Yes, as I put in the previous message, it is a hash created combining the hash of the input and the inner hash of the transaction, so it is not viable to obtain information from it.

I think that creating a new function is better than modifying the current one, so that previous versions of the software wallet can communicate with the new version of the firmware. The new function could receive the entire unsigned transaction (as a JSON string or a encoded string) o the lists of inputs and outputs, whichever is more convenient.

Senyoret1, 2019-01-24:
Great. I did some testing and found several errors, but conceptually the changes are what I had in mind, so I think it is going pretty good. Here are the details that I found:

- The most important thing is that the signature returned by the function is not correct. I'm not sure what the problem is, because I have not checked the C code (I'm going to take a look at it, but I do not know how far I’ll be able to go). When trying to inject the transaction this is the result:
[image]

- Although Base58 is a versatile format for converting byte arrays to strings, Steve said that it should only be used in the Skycoin code base to encode addresses, so the signatures should be returned as hex strings.

- If the transaction has a single input, the response is received correctly. However, if it has 2, the second signature is returned with one letter less and `\u0000` added at the end. If it has 3, the procedure fails with this error: `Wire format is invalid RangeError: index out of range: 2 + 89 > 17`

- When requesting confirmation from the user on the hardware wallet screen, “coin” and “hour” should be plural if the value is different from 1. Additionally, the values of the coins must be divided by 1,000,000, because right now the value is being shown in droplets, not coins.

- It would be a good usability improvement not to show the return address while asking for confirmation, because some users may not recognize it (or simply do not know the concept of return address) and be alarmed because it is different from the one they entered. For this to work, the output objects sent to the api could have an additional parameter with a name like `returnAddressIndex`, indicating the index of the address in case it is a return address. The firmware could check that parameter and, if it contains a valid index, check if that address is actually in the indicated index and, if so, ignore the output for the purposes of user confirmation. Confirming if the address really belongs to the wallet and if it is in the indicated index is important for security, so the speed of the check could be a factor to consider for implementing this suggestion. Also, maybe @gz-c would like to take a look at this.

- The parameters `nbIn` and `nbOut` of `devSkycoinTransactionSign` are not really necessary in Javascript, since you could read the `length` property of the arrays.

- Instead of writing `Pushing transaction with` into the console, it would be better to write `Pushing input with` and `Pushing output with`.

olemis, 2019-01-25:
The items in point 4 of [this comment](https://github.com/skycoin/hardware-wallet/issues/38#issuecomment-457395620) depend on the fiber coin params (droplet precision, ticker, coin and hours display name) . These should be sent from js (see #63 ) instead of been hard coded . Related to skycoin/skycoin#2117 .

mpsido, 2019-01-25:
@olemis @Senyoret1 I disagree the issue #47 is about something else. It talks about the python script performing the firmware signature. 
Right here we are talking about the transaction signatures, this does not go through that python script.

About the bad signature, I believe in this case the problem is not the signature but the computation of the actual message being signed.

In your previous message the gave me an example of innerHash, this was very helpful for me to reproduce that computation in the device.

## Missing step to compute message to be signed
Then I had a missing step between innerHash and message being signed: I found that missing step here: https://github.com/skycoin/skycoin/blob/develop/src/coin/transactions.go#L238

It is concatenating innerHash with transaction innerHash. That is what I had the device sign.
I will try to generate more examples using the file [transactions_test.go](https://github.com/skycoin/skycoin/blob/develop/src/coin/transactions_test.go). But if you see this message before I go through it      you may help me save some time.
This is what I meant when I said I want examples for my automatic test.

## Don't use base58 format for signature

I got it, but I think for now better fix things one by one. This format change will also affect the other signature message, it deserves an issue and a PR of its own.

## Messaging error
> * If the transaction has a single input, the response is received correctly. However, if it has 2, the second signature is returned with one letter less and `\u0000` added at the end. If it has 3, [...]
I will investigate this

## User message display
>     * When requesting confirmation from the user on the hardware wallet screen, “coin” and “hour” should be plural if the value is different from 1. Additionally, the values of the coins must be divided by 1,000,000, because right now the value is being shown in droplets, not coins.
> 

Ok, 

## returnAdresss

> * It would be a good usability improvement not to show the return address while asking for confirmation, because some users may not recognize it (or simply do not know the concept of return address) [...]

I don't understand what is a "return address" but let me try a guess: it is an address belonging to the wallet set in one of the output of the transaction. Am I right ?
If so, I think it would be excessive to have the embedded wallet check in its internal addresses if one of them matches one of the output (the addresses are not saved, but computed on the fly).
If that is important we can have an extra indication in the TransactionOutput structure to indicate if the output is an output address. What about this ?

## nbIn nbOut

>     * The parameters `nbIn` and `nbOut` of `devSkycoinTransactionSign` are not really necessary in Javascript, since you could read the `length` property of the arrays.
> 

I need these inputs on the C side, but yes I can test having them generated by the javascript code inside device-wallet.js

## Js log message

>     * Instead of writing `Pushing transaction with` into the console, it would be better to write `Pushing input with` and `Pushing output with`.

Ok, by the way about the device-wallet.js module, feel yourself home: change all the logs you want. I know I write some "bad english sentences" in my logs.

mpsido, 2019-01-25:
> ## Messaging error
>
>
> I will investigate this

Fixed [9b58ac19d06afb5a66efde014f704fc9e5f904e1](https://github.com/skycoin/hardware-wallet-js/pull/35/commits/9b58ac19d06afb5a66efde014f704fc9e5f904e1)

> ## User message display
>
>
> Ok,

Commit 44e21c7ae7a04312296cce088f2951bdc88baae8

> ## nbIn nbOut
>
>
> I need these inputs on the C side, but yes I can test having them generated by the javascript code inside device-wallet.js

Commit [90665caf559fb14cfab2cd72b5ce0b6004424729](https://github.com/skycoin/hardware-wallet-js/pull/35/commits/90665caf559fb14cfab2cd72b5ce0b6004424729)

> ## Js log message
>
>
> Ok, by the way about the device-wallet.js module, feel yourself home: change all the logs you want. I know I write some "bad english sentences" in my logs.

Commit [2815e9d0a34efe3d5925af513cceccc04de1deb2](https://github.com/skycoin/hardware-wallet-js/pull/35/commits/2815e9d0a34efe3d5925af513cceccc04de1deb2)

Senyoret1, 2019-01-25:
I am creating a code for generating data samples. I'll send you some examples after it is ready.

Senyoret1, 2019-01-26:
@mpsido 
As you said that the problem should be in the creation of the data that must be signed, I created some data for testing purposes. The data contains two parts, one with the JSON representation of the inputs and outputs of the transaction (the data is not exactly the same as the one expected by the hardware wallet, but it should work) and another with the inner hash that the transaction should have and the list of values that should be signed. As I do not know what format you find most comfortable to use, the data is in hex and Base58.

The transaction data were created manually, so it is not necessarily real, but the expected data was calculated by the same Go code that the web wallet uses to create transactions. To check if the code that generates the expected data was configured correctly, I used it to create a transaction with data from a test wallet, which I then signed and could send to the blockchain without problems, so the data should be correct.

I added 10 examples and tried to make them diverse (one or several inputs, one or several outputs, inputs with and without hours, input and output with the same address, etc.). If you need anything else, let me know.

[test_data-.txt](https://github.com/skycoin/hardware-wallet/files/2798360/test_data-.txt)

olemis, 2019-01-26:
@mpsido @Senyoret1 maybe [cipher golden files](https://github.com/skycoin/skycoin/tree/develop/src/cipher/testsuite/testdata) could be of some help as well .

mpsido, 2019-01-26:
> Yes, the return address is that. More specifically, when a transaction is created and the inputs have more coins than the amount the user wants to send, the software wallet selects an address from the wallet and creates and additional output to send the remaining coins to it. Basically it is what allows the user to send arbitrary amounts of coins without having to have inputs with the exact amounts or lose coins.
> 
> Taking that into account, it is not necessary to check the addresses of all the outputs to know which ones belongs to the wallet (in fact more than one could belong to the wallet if the user is moving coins from one address to another), but the output used to return coins must be identified as such. The thing is that, as you say, I could indicate in the `TransactionOutput` structure if the output is just being used to return coins and that it should not be show while asking for confirmation, but the hardware wallet should not trust that value, because a hacker could add an additional output for sending coins to himself and hide it from the confirmation screen simply by saying that it is an output for returning coins. The hardware wallet should check if the output is in fact returning coins to the user wallet.
> 
> We can assume that the software wallet will only include one output for returning coins, and indicate which one is it when calling the function for creating transaction. However, if there are still performance problems, we could also establish as a condition that the output for returning coins should always send the coins to the first address in the wallet, since it is the fastest to process.

Here is what I suggest: for every output address that is a 'return address' insert in the TransactionOutput the index of the address in the wallet (the same index I would pass to the addressGen function to generate that same address).
I can work easily with it. just have the hardware wallet generate the address at that index and check they match

mpsido, 2019-01-26:
> @mpsido
> As you said that the problem should be in the creation of the data that must be signed, I created some data for testing purposes. The data contains two parts, one with the JSON [...]

Thank you so much @Senyoret1 I think I can work with that. I'll come back to you with a better code.

Senyoret1, 2019-01-26:
> Here is what I suggest: for every output address that is a 'return address' insert in the TransactionOutput the index of the address in the wallet (the same index I would pass to the addressGen [...]

Great, that sounds good

mpsido, 2019-01-26:
@Senyoret1 

## All test pass
I used your test cases in the latest commit: 8c91f8fff65f12a0dee6b6e951d2ca5899b4f974 and they all pass.
So then I am surprised the signature is not valid. 

## Signatures are valid on my side
I did also check the output signatures with the device's check signature function and the address matches

```
 go run cli.go emulatorCheckMessageSignature --message=d9ea31ca8b461e7dc9879916cd71631dd118aee78697aa00c7db0e8b32256f91 --signature=JH3MgWGJ7Bz4o4baSEZnC8ZAUEzjnmHeWrnwhRrspjfFHuXVmQ8Uj7Fpn8CgDLtKfx4kCiPpmCqdRiQQvgGx4xQdv
```

## One bug when we have two inputs (not 1 not 3)

I did notice a seemingly not related bug. When I have one input the signatures are sent properly, same when I have 3 inputs. But weirdly enough when I have 2, the last character of the second signature is truncated:
I receive something like this: 
```
[ 'FRywBB8cJu5vqT7NuRuAbmwGnueroiNGyaAccjbfSuo2Bz8dyq3fXXrFRJ3wEcZEN4EercC98fpSCc74amxujPKAY',
 '52RjHr4hZd5cBYensHBR87e8bfTX5VrAH2A95EaaNGF7uK7iYzwZ87yiwYyrUc5s21uFKyswAhxohWvM4j5w6KJ8\u0000' ]
```
I have an idea and I am still investigating, this has to do with a sort of edge effect. We receive 3 chunks of 64 bytes and that last letter happen to be in the very last byte of the chunk.

Can you check again the validity of the signatures the device outputs do you still have the error "Transaction violates hard constraints" ?

mpsido, 2019-01-26:
I believe the issue is in the protobuf  unmarshaling function in commit 53f1c2d7aded6aaee8eb8eb648af401648c5d168 I found a simple hack to make sure that situation won't happen to us

mpsido, 2019-01-26:
Passed optional parameter to the device to detect "return value"

Commit b91d286d34df34a579227c47f504fe957b2b2fe3 and on its corresponding commit on JS API (here)[https://github.com/skycoin/hardware-wallet-js/pull/35/commits/298731177131166226bd18d0b2a0fe4b1a3723b4].

mpsido, 2019-01-28:
@Senyoret1 
It works on my side:

I pushed the test code in https://github.com/skycoin/hardware-wallet-js/blob/transactions/main.js so we have the same code, (see the log output below).

The one thing I noticed that may have caused you the error is the SetMnemonic function resolves without error if the operation was refused because of mnemonic already set (code here)[https://github.com/skycoin/hardware-wallet-js/blob/transactions/device-wallet.js#L1113]. It appears in the log below.
In order to fix that I also pushed a modification that make the function reject the promise in that case.
And I added a Wipe instruction in the test as well.
Tell me if that works better now.

```
dev 46d / 4052 : usage_page:de88, usage:a995
dev 46d / c52b : usage_page:de88, usage:a995
Skycoin hardware NOT FOUND, using emulator
Device Open
Sending data <Buffer 3f 23 23 00 71 00 00 00 47 0a 45 63 6c 6f 75 64 20 66 6c 6f 77 65 72 20 75 70 73 65 74 20 72 65 6d 61 69 6e 20 67 72 65 65 6e 20 6d 65 74 61 6c 20 62 ... > 128
Sending data 64
Sending data 64
server got: 
                        ?##.Device is already initialized. Use Wipe first. from 127.0.0.1:21324
Received data of length:  64
That message says its size is:  50
msgid slice: Uint8Array [ 0, 3 ]
Slice: Uint8Array [ 0, 0, 0, 50 ]
Received header  msg this.kind:  MessageType_Failure  size:  50 buffer lenght:  64 
Remaining bytesToGet: -5
Device Close
Skiping button confirmation! MessageType_Failure
Device Open
Sending data <Buffer 3f 23 23 00 72 00 00 00 04 0a 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... > 64
Sending data 64
server got: 
                        ?##u%
#2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw from 127.0.0.1:21324
Received data of length:  64
That message says its size is:  37
msgid slice: Uint8Array [ 0, 117 ]
Slice: Uint8Array [ 0, 0, 0, 37 ]
Received header  msg this.kind:  MessageType_ResponseSkycoinAddress  size:  37 buffer lenght:  64 
Remaining bytesToGet: -18
Device Close
Addresses generation received message kind:  MessageType_ResponseSkycoinAddress
Addresses [ '2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw' ]
Address Promise resolved [ '2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw' ]
Pushing input: a885343cc57aedaab56ad88d860f2bd436289b0248d1adc55bcfa0d9b9b807c3 index:  0
Pushing output: zC8GAQGQBfwk7vtTxVoRG7iMperHNuyYPs coin: 1000000 hour: 1 address_index: undefined
ArrayBuffer input len:  1
ArrayBuffer output len:  1
Device Open
Sending data <Buffer 3f 23 23 00 7a 00 00 00 76 0a 01 12 44 0a 40 61 38 38 35 33 34 33 63 63 35 37 61 65 64 61 61 62 35 36 61 64 38 38 64 38 36 30 66 32 62 64 34 33 36 32 ... > 192
Sending data 64
Sending data 64
Sending data 64
server got: 
                        ?## from 127.0.0.1:21324
Received data of length:  64
That message says its size is:  2
msgid slice: Uint8Array [ 0, 26 ]
Slice: Uint8Array [ 0, 0, 0, 2 ]
Received header  msg this.kind:  MessageType_ButtonRequest  size:  2 buffer lenght:  64 
Remaining bytesToGet: -53
Device Close
TransactionSign received message kind: MessageType_ButtonRequest
ButtonRequest!
Device Open
Sending data <Buffer 3f 23 23 00 1b 00 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... > 64
Sending data 64
server got: 
                        ?## from 127.0.0.1:21324
Received data of length:  64
That message says its size is:  2
msgid slice: Uint8Array [ 0, 26 ]
Slice: Uint8Array [ 0, 0, 0, 2 ]
Received header  msg this.kind:  MessageType_ButtonRequest  size:  2 buffer lenght:  64 
Remaining bytesToGet: -53
User hit a button
Device Close
TransactionSign received message kind: MessageType_ButtonRequest
ButtonRequest!
Device Open
Sending data <Buffer 3f 23 23 00 1b 00 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... > 64
Sending data 64
server got: 
                        ?##{]
YB5jgJMUXM4V5FUj7LYXAGQu6fCaR4xeHZQ3hHHzrJRKFYmZ1YGYyV from 127.0.0.1:21324
Received data of length:  64
That message says its size is:  93
msgid slice: Uint8Array [ 0, 123 ]
Slice: Uint8Array [ 0, 0, 0, 93 ]
Received header  msg this.kind:  MessageType_ResponseTransactionSign  size:  93 buffer lenght:  128 
Remaining bytesToGet: 38
server got: 
                        ?E5mCR7Qn8f2r3jpRCrvLE5VSByDeHBh1381u from 127.0.0.1:21324
Received data  msg kind:  MessageType_ResponseTransactionSign  size:  93 buffer lenght:  128 
Remaining bytesToGet: -26
User hit a button
Device Close
TransactionSign received message kind: MessageType_ResponseTransactionSign
Uint8Array [ 56, 49, 117, 16, 0 ] 93
Signature promise resolved [ 'B5jgJMUXM4V5FUj7LYXAGQu6fCaR4xeHZQ3hHHzrJRKFYmZ1YGYyVE5mCR7Qn8f2r3jpRCrvLE5VSByDeHBh1381u' ]
Device Open
Sending data <Buffer 3f 23 23 00 73 00 00 00 c2 0a 23 32 45 55 33 4a 62 76 65 48 64 6b 78 57 36 7a 35 74 64 68 62 62 42 32 6b 52 41 57 76 58 43 32 70 4c 7a 77 12 40 32 37 ... > 256
Sending data 64
Sending data 64
Sending data 64
Sending data 64
server got: 
                        ?##%
#2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw from 127.0.0.1:21324
Received data of length:  64
That message says its size is:  37
msgid slice: Uint8Array [ 0, 2 ]
Slice: Uint8Array [ 0, 0, 0, 37 ]
Received header  msg this.kind:  MessageType_Success  size:  37 buffer lenght:  64 
Remaining bytesToGet: -18
Device Close
Address emiting that signature: 2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw
```

Senyoret1, 2019-01-28:
I tried again, this time with `main.js`, and the error continues. I also tried with Linux, to check if it is an OS realted error, but same problem. However, as the output you posted was generated using the emulator, I tested using the emulator and it worked correctly, so the problem seems to occur only while using a physical device.

Just to be sure, I recompiled the firmware, installed it with the JS library and tried again, but the problem continued. I also tried creating a combined build (firmware + bootloader) and installing it with a ST Link device, but the result is the same.

mpsido, 2019-01-29:
@Senyoret1 that surprises me because I also tested with the device. I am gonna need your logs

Senyoret1, 2019-01-29:
Wow, then it's a weird problem. Here is the console output:
```
$ node main.js
Skycoin hardware is plugged in
Device Open
Writing a buffer of length  64 to the device
Received data of length:  64
That message says its size is:  2
msgid slice: Uint8Array [ 0, 26 ]
Slice: Uint8Array [ 0, 0, 0, 2 ]
Received header  msg this.kind:  MessageType_ButtonRequest  size:  2 buffer lenght:  64
Remaining bytesToGet: -53
Device Close
ButtonRequest!
Device Open
Writing a buffer of length  64 to the device
Received data of length:  64
That message says its size is:  14
msgid slice: Uint8Array [ 0, 2 ]
Slice: Uint8Array [ 0, 0, 0, 14 ]
Received header  msg this.kind:  MessageType_Success  size:  14 buffer lenght:  64
Remaining bytesToGet: -41
User hit a button
Device Close
Device Open
Writing a buffer of length  128 to the device
Received data of length:  64
That message says its size is:  2
msgid slice: Uint8Array [ 0, 26 ]
Slice: Uint8Array [ 0, 0, 0, 2 ]
Received header  msg this.kind:  MessageType_ButtonRequest  size:  2 buffer lenght:  64
Remaining bytesToGet: -53
Device Close
ButtonRequest!
Device Open
Writing a buffer of length  64 to the device
Received data of length:  64
That message says its size is:  71
msgid slice: Uint8Array [ 0, 2 ]
Slice: Uint8Array [ 0, 0, 0, 71 ]
Received header  msg this.kind:  MessageType_Success  size:  71 buffer lenght:  128
Remaining bytesToGet: 16
Reading one more time HID {
  domain: null,
  _events: { newListener: [Function] },
  _eventsCount: 1,
  _maxListeners: undefined,
  _raw: HID {},
  write: [Function: bound write],
  getFeatureReport: [Function: bound getFeatureReport],
  sendFeatureReport: [Function: bound sendFeatureReport],
  setNonBlocking: [Function: bound setNonBlocking],
  readSync: [Function: bound readSync],
  readTimeout: [Function: bound readTimeout],
  getDeviceInfo: [Function: bound getDeviceInfo],
  _paused: true }
Received data  msg kind:  MessageType_Success  size:  71 buffer lenght:  128
Remaining bytesToGet: -48
User hit a button
Device Close
Device Open
Writing a buffer of length  64 to the device
Received data of length:  64
That message says its size is:  37
msgid slice: Uint8Array [ 0, 117 ]
Slice: Uint8Array [ 0, 0, 0, 37 ]
Received header  msg this.kind:  MessageType_ResponseSkycoinAddress  size:  37 buffer lenght:  64
Remaining bytesToGet: -18
Device Close
Addresses generation received message kind:  MessageType_ResponseSkycoinAddress
Addresses [ '2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw' ]
Address Promise resolved [ '2EU3JbveHdkxW6z5tdhbbB2kRAWvXC2pLzw' ]
Pushing input: a885343cc57aedaab56ad88d860f2bd436289b0248d1adc55bcfa0d9b9b807c3 index:  0
Pushing output: zC8GAQGQBfwk7vtTxVoRG7iMperHNuyYPs coin: 1000000 hour: 1 address_index: undefined
ArrayBuffer input len:  1
ArrayBuffer output len:  1
Device Open
Writing a buffer of length  192 to the device
Received data of length:  64
That message says its size is:  2
msgid slice: Uint8Array [ 0, 26 ]
Slice: Uint8Array [ 0, 0, 0, 2 ]
Received header  msg this.kind:  MessageType_ButtonRequest  size:  2 buffer lenght:  64
Remaining bytesToGet: -53
Device Close
TransactionSign received message kind: MessageType_ButtonRequest
ButtonRequest!
Device Open
Writing a buffer of length  64 to the device
Received data of length:  64
That message says its size is:  2
msgid slice: Uint8Array [ 0, 26 ]
Slice: Uint8Array [ 0, 0, 0, 2 ]
Received header  msg this.kind:  MessageType_ButtonRequest  size:  2 buffer lenght:  64
Remaining bytesToGet: -53
User hit a button
Device Close
TransactionSign received message kind: MessageType_ButtonRequest
ButtonRequest!
Device Open
Writing a buffer of length  64 to the device
Received data of length:  64
That message says its size is:  93
msgid slice: Uint8Array [ 0, 123 ]
Slice: Uint8Array [ 0, 0, 0, 93 ]
Received header  msg this.kind:  MessageType_ResponseTransactionSign  size:  93 buffer lenght:  128
Remaining bytesToGet: 38
Reading one more time HID {
  domain: null,
  _events: { newListener: [Function] },
  _eventsCount: 1,
  _maxListeners: undefined,
  _raw: HID {},
  write: [Function: bound write],
  getFeatureReport: [Function: bound getFeatureReport],
  sendFeatureReport: [Function: bound sendFeatureReport],
  setNonBlocking: [Function: bound setNonBlocking],
  readSync: [Function: bound readSync],
  readTimeout: [Function: bound readTimeout],
  getDeviceInfo: [Function: bound getDeviceInfo],
  _paused: true }
Received data  msg kind:  MessageType_ResponseTransactionSign  size:  93 buffer lenght:  128
Remaining bytesToGet: -26
User hit a button
Device Close
TransactionSign received message kind: MessageType_ResponseTransactionSign
Uint8Array [ 101, 89, 55, 16, 0 ] 93
Signature promise resolved [ 'GvKS4S3CA2YTpEPFA47yFdC5CP3y3qB18jwiX1URXqWPvJw3iFSLeSntn1xxaXovMUdH6qoicHWauNDdKYtggXeY7' ]
Device Open
Writing a buffer of length  256 to the device
Received data of length:  64
That message says its size is:  36
msgid slice: Uint8Array [ 0, 2 ]
Slice: Uint8Array [ 0, 0, 0, 36 ]
Received header  msg this.kind:  MessageType_Success  size:  36 buffer lenght:  64
Remaining bytesToGet: -19
Device Close
Promise rejected Error: Wrong signature
```

Senyoret1, 2019-01-29:
Taking into account that the results do not match, it would be good if someone else could perform the test. @olemis , would it be possible for you or someone else to perform the test? What should be done is compile the changes on the `transactions` branch of this repo, install the firmware in an hardware wallet, run `node main.js` in the JS library folder (also using the `transactions` branch of that repository), and accept everything in the hardware wallet. The objective is to see in the console at the end of the process if the signature generated by the hardware wallet is valid or not.

olemis, 2019-01-29:
@Senyoret1 sure ... in advance I tell you , the signature might not be valid for the reasons mentioned in #47

stdevAlDen, 2019-01-29:
> Taking into account that the results do not match, it would be good if someone else could perform the test. @olemis , would it be possible for you or someone else to perform the test? What should be [...]

I had have the same error, I'm going to look in deep for more details, I'll keep you updated for any news.
In the emulator it works ok.
PD:
- v9.11.1
- Linux ciaamq 4.19.0-1-amd64 #1 SMP Debian 4.19.12-1 (2018-12-22) x86_64 GNU/Linux
- arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/7.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:7-2018-q2-4 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' LDFLAGS=-Wl,-z,relro OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-9JP9oM/gcc-arm-none-eabi-7-2018-q2=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (15:7-2018-q2-4)
- arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/opt/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: /tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/src/gcc/configure --target=arm-none-eabi --prefix=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native --libexecdir=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native/lib --infodir=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/install-native/arm-none-eabi --build=x86_64-linux-gnu --host=x86_64-linux-gnu --with-gmp=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/build-native/host-libs/usr --with-mpfr=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/build-native/host-libs/usr --with-mpc=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/build-native/host-libs/usr --with-isl=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/build-native/host-libs/usr --with-libelf=/tmp/jenkins-GCC-6-buildandreg-223_20170621_1498033910/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors 6-2017-q2-update' --with-multilib-list=rmprofile
Thread model: single
gcc version 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437] (GNU Tools for ARM Embedded Processors 6-2017-q2-update)

mpsido, 2019-01-29:
I reproduced it

Senyoret1, 2019-01-29:
Great, thanks @olemis, @stdevAlDen and @mpsido. This is an important step in the right direction.

mpsido, 2019-01-30:
@Senyoret1  @stdevAlDen  I just pushed a change that fixed the issue I reproduced.
Wish it works for you this time

stdevAlDen, 2019-01-30:
> @Senyoret1 @stdevAlDen I just pushed a change that fixed the issue I reproduced.
> Wish it works for you this time

Great work @mpsido I just going to test for feed back.

stdevAlDen, 2019-01-30:
> I did some in-depth testing and everything worked correctly, I was able to inject into the blockchain transactions with different numbers of outputs and inputs. I think I would only change 3 small details:
> 
>     * The hardware wallet screen does not show the decimal part of the coin values. For example, instead of showing “1.2”, it simply shows “1”.
> 
>     * To follow the standard used in other Skycoin wallets, while asking for confirmation it would be good to show the coins in the top line and the hours in the bottom line, not the opposite, as is being done now. I mean just swaping the two lines.
> 
>     * Currently when trying to send 1 coin or 1 hour, the hardware wallet shows “1 coins” and/or “1 hours”. In that case it would be good to show “1 coin” and “1 hour”, in singular. I mean something like `coins == 1 ¿ 'coin' : 'coins'`.

Yes @mpsido this worked for me too, no a in deep test but the basic one worked :smile:

mpsido, 2019-01-31:
> I did some in-depth testing and everything worked correctly, I was able to inject into the blockchain transactions with different numbers of outputs and inputs. I think I would only change 3 small [...]

Done

olemis, 2019-02-07:
fixed by bbbf1cf

PR #62 (2019-02-07): Transactions-C-firmware
Fixes #38 
Works with JS code on PR skycoin/hardware-wallet-js#35
closed 2019-02-07 by olemis