skycoin/hardware-wallet #45: [Hardware-Wallet] How to figure out if MEMORY_PROTECT was enabled

mpsido, 2019-01-11:


mpsido, 2019-01-11:
@gz-c :   Can the bootloader affect the display to visually indicate whether or not memory protection is on?

mpsido, 2019-01-17:
Use the compilation flag activating the memory protection to declare some code saying "I am not memory protected" if not defined `#ifndef MEMORY_PROTECT`...etc...

mpsido, 2019-01-18:
Read more about memory protection here: https://www.st.com/content/ccc/resource/technical/document/application_note/89/12/c5/e2/0d/0e/45/7f/DM00186528.pdf/files/DM00186528.pdf/jcr:content/translations/en.DM00186528.pdf

mpsido, 2019-01-21:
Solution is to create a message reception procedure in the bootloader for GetVersion message code:

This is the buffer sent to the device from the golang code if we send the GetVersion message with memoryProtect option

[getVersionMemprotect.patch.txt](https://github.com/skycoin/hardware-wallet/files/2777480/getVersionMemprotect.patch.txt)

2019/01/21 11:31:05 [[63 35 35 0 120 0 0 0 2 10 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]]

jdknives, 2019-03-09:
@olemis is it possible to make the bootloader invert the Skycoin logo if not memory protected? Have we implemented binary labeling to indicate mem_protect status and version?

olemis, 2019-04-09:
> is it possible to make the bootloader invert the Skycoin logo if not memory protected?

yes

> Have we implemented binary labeling to indicate mem_protect status and version?

there are versioning rules but they do not consider mem protect since every final product should be memory protected prior to flashing ... gotta give it a bit more thought

stdevAlDen, 2019-04-18:
So far now I was not able to read the RDP level from the firmware, only from the bootloader, it is not useful to do it in the bootloader, because if you do not enter in bootloader mode, you will not see it. An alternative to enforce the `MEMORY_PROTECTION` to be enabled is check the value at boot time and if the value is not `2` abort the boot and show a screen message informing about the problem

stdevAlDen, 2019-04-18:
I'm trying to figure out how [`memory_protection_enabled`](https://github.com/skycoin/hardware-wallet/issues/45#issuecomment-455937763) but it's difficult if you look at the specification:
[image]

so a simple check over the bits in range `15:8` in the `FLASH_OPTCR` should be enough, but it's required to think in this to avoid a mistake, maybe I'm missing something

stdevAlDen, 2019-04-18:
I think that there is a bug here:
https://github.com/skycoin/hardware-wallet/blob/384c508d07b8909c58f9afba5b8677fc409e66bb/tiny-firmware/memory.c#L48
as this line is setting the `FLASH_OPTCR` register as a whole and maybe setting the bit in `15:8` range should be enough to avoid overwriting other flags by accident.

stdevAlDen, 2019-04-20:
> I'm trying to figure out how [`memory_protection_enabled`](https://github.com/skycoin/hardware-wallet/issues/45#issuecomment-455937763) but it's difficult if you look at the specification:
> [image] [...]

I'm having problems to read the `FLASH_OPTCR` register, the `hard_fault_handler` is getting triggered, the code is:
```c
uint8_t memory_protection_level(void) {
	uint32_t *opts_register = (uint32_t*)FLASH_OPTCR;
	uint32_t mask = 0B00000000000000001111111100000000;
	switch (((*opts_register) & mask) >> 8) {
		case 0xAA:
			return 0;
		case 0xCC:
			return 2;
		default:
			return 1;
	}
}
```

Looking at a different example where the library is st peripherals(different library but the important detai is the use of `HAL_FLASH_OB_Unlock`):
```c
void lora_flash_read_protection_check( void ) {
   /* Unlock the Flash to enable the flash control register access *************/
   HAL_FLASH_Unlock();
   /* Unlock the Options Bytes *************************************************/
   HAL_FLASH_OB_Unlock();
   FLASH_OBProgramInitTypeDef OptionsBytesStruct;
   /* Get pages write protection status ****************************************/
   HAL_FLASHEx_OBGetConfig(&OptionsBytesStruct);
   switch( OptionsBytesStruct.RDPLevel ) {
      case OB_RDP_LEVEL_0: PRINTF('RDPLevel : LEVEL 0 \r\n' ); break;
      case OB_RDP_LEVEL_1: PRINTF('RDPLevel : LEVEL 1 \r\n' ); break;
      case OB_RDP_LEVEL_2: PRINTF('RDPLevel : LEVEL 2 \r\n' ); break;
   }
   uint8_t bRDPSet = false;
   /* Check if readoutprtection is enabled ***********************/
   if((OptionsBytesStruct.RDPLevel) == OB_RDP_LEVEL_0)
   {
      OptionsBytesStruct.OptionType= OPTIONBYTE_RDP;
      OptionsBytesStruct.RDPLevel = OB_RDP_LEVEL_1;
      if(HAL_FLASHEx_OBProgram(&OptionsBytesStruct) != HAL_OK) {
         PRINTF('Fail to Read-Out Protection Settings! \r\n');
      } else
      bRDPSet = true;
      /* Generate System Reset to load the new option byte values ***************/
      HAL_FLASH_OB_Launch();
   }
   /* Lock the Options Bytes *************************************************/
   HAL_FLASH_OB_Lock();
   HAL_FLASH_Lock();
}
```

stdevAlDen, 2019-04-22:
@olemis very important, security related
https://community.st.com/s/question/0D50X00009Xke7aSAB/readout-protection-cracked-on-stm32
https://www.aisec.fraunhofer.de/en/FirmwareProtection.html

stdevAlDen, 2019-04-22:
> I'm having problems to read the `FLASH_OPTCR` register, the `hard_fault_handler` is getting triggered, the code is:
>
> ```c
> uint8_t memory_protection_level(void) {
> 	uint32_t *opts_register = [...]

the problem look to be related to a required call to:
```
	flash_unlock();
	flash_unlock_option_bytes();
```
but it's not working either.

stdevAlDen, 2019-04-22:
> @olemis very important, security related
> https://community.st.com/s/question/0D50X00009Xke7aSAB/readout-protection-cracked-on-stm32
> https://www.aisec.fraunhofer.de/en/FirmwareProtection.html

@olemis https://github.com/trezor/trezor-firmware/blob/b4cec0ae541adca38ccfc46ffe6ec6ff0a3038fb/legacy/memory.c#L62

olemis, 2019-05-03:
fixed in 95f43f7

PR #200 (2019-05-03): Get memprotect level ref #45
Fixes #45
closes #187 

Changes:
- ask the device for the `RDP` level: `rdp_level`
- `send_msg_features` have a new field ("rdp level info") and changes bitcointrezor.com to skycoin.net
- invert logo if red level is different to `2`.

Does this change need to mentioned in CHANGELOG.md?
yes

Requires changes in protobuff specs?
yes
https://github.com/skycoin/hardware-wallet-protob/pull/49
in bootloader too, in this delta

Requires testing
yes, I need someone to help me to test this feature in a physical device but putting rdp level to 2 will **make the device unable [... cut]
closed 2019-05-03 by olemis