skycoin/hardware-wallet #47: [Hardware-Wallet] sometimes firmware-signature.py fails

mpsido, 2019-01-11:
Firmware fingerprint: 212223fdd8e0e36249eb303b1d23e2c608be4a6be3a1c736486da0166fa7d7be
Skycoin signature: (censured)
Traceback (most recent call last):
  File "tiny-firmware/bootloader/firmware_sign.py", line 178, in <module>
    main(args)
  File "tiny-firmware/bootloader/firmware_sign.py", line 169, in main
    data = sign(data)
  File "tiny-firmware/bootloader/firmware_sign.py", line 145, in sign
    raise Exception("Signature lenght {} is not correct".format(len(signature.value)))
Exception: Signature lenght 32 is not correct
make: *** [sign] Error 1

gz-c, 2019-01-11:
This happens for some signatures, despite the reported signature being a valid 64 character hex encoded string.

mpsido, 2019-01-14:
The error is a "zero" padding problem. 
The string is expected to be 64 characters long, but if there is a zero byte somewhere, it is cut short. Indeed the string is a "C like char" because the underlining library performing the signature is the .so file generated from the code under [skycoin-api](https://github.com/skycoin/hardware-wallet/tree/master/skycoin-api) folder.

It could be fixed by dealing with the different cases, but it is not "harmful" and can be fixed by just repeating the process again.

As a matter of fact there is already a loop in the code [firmware_sign.py#L138](https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/bootloader/firmware_sign.py#L138) trying to sign ten times until the buffer do not contain zeros.

In short: that looks like a sort of "proof of work" finding a hash not containing any zero bytes.

It is not harmful so not prioritary to fix this.

stdevAlDen, 2019-01-25:
Correct me if I am wrong but I think that [this kind of signature](https://github.com/skycoin/hardware-wallet/issues/47#issuecomment-453903791) have to work as it's works in [libskycoin](https://github.com/skycoin/libskycoin/) project, there is no thing like POW in skycoin, it's an issue. 

For each [skycoin-api](https://github.com/skycoin/hardware-wallet/tree/master/skycoin-api) function I have be able find one equivalent in libskycoin and I can't:

```{.c}

// extern GoUint32 SKY_cipher_PubKeyFromSig(cipher__Sig* p0, cipher__SHA256* p1, cipher__PubKey* p2);
int recover_pubkey_from_signed_message(const char* message, const uint8_t* signature, uint8_t* pubkey);

void ecdsa_get_public_key33(const ecdsa_curve *curve, const uint8_t *priv_key, uint8_t *pub_key);
// const curve_info *get_curve_by_name(const char *curve_name);
// int hdnode_from_seed(const uint8_t *seed, int seed_len, const char* curve, HDNode *out);
// void hdnode_fill_public_key(HDNode *node);
void create_node(const char* seed_str, HNode* node);
void uncompress_mcoords(const ecdsa_curve *curve, uint8_t odd, const bignum256 *x, bignum256 *y);
int mecdsa_validate_pubkey(const ecdsa_curve *curve, const curve_point *pub);
void mpoint_multiply(const ecdsa_curve *curve, const bignum256 *k, const curve_point *p, curve_point *res);
void mscalar_multiply(const ecdsa_curve *curve, const bignum256 *k, curve_point *res);
void mpoint_set_infinity(curve_point *p);
int mpoint_is_infinity(const curve_point *p);
void mpoint_add(const ecdsa_curve *curve, const curve_point *cp1, curve_point *cp2);
// extern GoUint32 SKY_handle_copy(Handle p0, Handle* p1);
void mpoint_copy(const curve_point *cp1, curve_point *cp2);
int mpoint_is_equal(const curve_point *p, const curve_point *q);
int mpoint_is_negative_of(const curve_point *p, const curve_point *q);
// extern GoUint32 SKY_cipher_DoubleSHA256(GoSlice p0, cipher__SHA256* p1);
void mpoint_double(const ecdsa_curve *curve, curve_point *cp);
void mcurve_to_jacobian(const curve_point *p, jacobian_curve_point *jp, const bignum256 *prime);
void mjacobian_to_curve(const jacobian_curve_point *jp, curve_point *p, const bignum256 *prime);
void mpoint_jacobian_add(const curve_point *p1, jacobian_curve_point *p2, const ecdsa_curve *curve);
void mconditional_negate(uint32_t cond, bignum256 *a, const bignum256 *prime);
void mpoint_jacobian_double(jacobian_curve_point *p, const ecdsa_curve *curve);

// extern GoUint32 SKY_cipher_ECDH(cipher__PubKey* p0, cipher__SecKey* p1, coin__UxArray* p2);
void ecdh(const uint8_t* secret_key, const uint8_t* remote_public_key, uint8_t* ecdh_key /*should be size SHA256_DIGEST_LENGTH*/);
void ecdh_shared_secret(const uint8_t* secret_key, const uint8_t* remote_public_key, uint8_t* shared_secret /*should be size SHA256_DIGEST_LENGTH*/);
// extern GoUint32 SKY_cipher_SHA256_Set(cipher__SHA256* p0, GoSlice p1);
void secp256k1Hash(const uint8_t* seed, const size_t seed_length, uint8_t* secp256k1Hash_digest);
void generate_deterministic_key_pair_iterator(const uint8_t* seed, const size_t seed_length, uint8_t* nextSeed, uint8_t* seckey, uint8_t* pubkey);
// extern GoUint32 SKY_cipher_SumSHA256(GoSlice p0, cipher__SHA256* p1);
void compute_sha256sum(const uint8_t *seed, uint8_t* digest /*size SHA256_DIGEST_LENGTH*/, size_t seed_lenght);
// extern GoUint32 SKY_cipher_PubKeyFromSecKey(cipher__SecKey* p0, cipher__PubKey* p1);
void generate_pubkey_from_seckey(const uint8_t* seckey, uint8_t* pubkey);
// extern GoUint32 SKY_cipher_GenerateDeterministicKeyPair(GoSlice p0, cipher__PubKey* p1, cipher__SecKey* p2);
void generate_deterministic_key_pair(const uint8_t* seed, const size_t seed_length, uint8_t* seckey, uint8_t* pubkey);
// extern GoUint32 SKY_cipher_AddressFromPubKey(cipher__PubKey* p0, cipher__Address* p1);
void generate_base58_address_from_pubkey(const uint8_t* pubkey, char* address, size_t *size_address);
// extern void SKY_cipher_BitcoinAddressFromPubKey(cipher__PubKey* p0, cipher__BitcoinAddress* p1);
void generate_bitcoin_address_from_pubkey(const uint8_t* pubkey, char* address, size_t *size_address);
// extern GoUint32 SKY_cipher_BitcoinAddressFromSecKey(cipher__SecKey* p0, cipher__BitcoinAddress* p1);
void generate_bitcoin_private_address_from_seckey(const uint8_t* pubkey, char* address, size_t *size_address);
int ecdsa_skycoin_sign(const uint32_t nonce_value, const uint8_t *priv_key, const uint8_t *digest, uint8_t *sig);
// extern GoUint32 SKY_base58_String2Hex(GoString p0, coin__UxArray* p1);
void tohex(char * str, const uint8_t* buffer, int bufferLength);
// extern GoUint32 SKY_cipher_SecKeyFromHex(GoString p0, cipher__SecKey* p1);
void writebuf_fromhexstr(const char *str, uint8_t* buf);
```

A valuable feature could be the possibility of running [(a subset of) these tests](https://github.com/skycoin/libskycoin/tree/develop/lib/cgo/tests) against [skycoin-api](https://github.com/skycoin/hardware-wallet/tree/master/skycoin-api), so I am trying to figure out how to make this two APIs to match but it result very difficult for me, maybe I am wrong at all or I am missing some little details about the project but I would be very grateful that someone could help me understand.

stdevAlDen, 2019-01-27:
Hello, I just write a simple test:
```{.c}
    // https://github.com/simelo/skycoin/blob/develop/src/cipher/testsuite/testdata/seed-0000.golden#L2
    char seed[] = "aG9iYnkgZ3JhY2UgbHVtYmVyIGN1cnZlIGRvdWJsZSBzY29ycGlvbiByb3RhdGUgaW5zcGlyZSBzaGl2ZXIgYWRkcmVzcyB3ZWFwb24gZG9vcg";
    uint8_t hash[SHA256_DIGEST_LENGTH] = {0};
    compute_sha256sum((const uint8_t*)seed, hash, sizeof(seed));
    uint8_t seckey[32] = {0};
    // https://github.com/simelo/skycoin/blob/develop/src/cipher/testsuite/testdata/seed-0000.golden#L7
    memcpy(seckey, fromhex("6614f0956f4e08e07ec2d4c2767b9f7121449fc2269e2b22f1a549d63e9f719b"),  sizeof(seckey));
    uint8_t signature[65] = {0};
    ecdsa_skycoin_sign(1, seckey, hash, signature);
    printf("\n");
    char out[sizeof(signature) * 2];
    tohex(out, signature, sizeof(signature));
    for (unsigned long i = 0; i < sizeof(out); ++i) printf("%c", out[i]);
    printf("\n");
```
the output is: `79be667ef9dcbbac55a06295ce870b07029bfcdb2dce28d959f2815b16f817981140f989552d869d1c7a693bc787e8ad5c8ae9227ba7bb580086db7f62afdcd101`

The result look as a [deterministic implementation](https://security.stackexchange.com/a/46943), but in [libskycoin](http://github.com/skycoin/libskycoin/) this look random, in addition the result should be expected one of the listed [here](https://github.com/simelo/skycoin/blob/develop/src/cipher/testsuite/testdata/seed-0000.golden#L9)?

My major question is if is required the actual difference that you can found between the libskycoin functions signature and the hardware-wallet/skycoin-api functions signature, some one can help me to understand?

mpsido, 2019-01-27:
one of the parameters of edcsa_skycoin_sign function is a number sort of nonce that can be random when called in fsm.c this function gets a random integer as input.

that is the best way I found to make this function testable but still allow some randomness in the signature

stdevAlDen, 2019-05-02:
> Correct me if I am wrong but I think that [this kind of signature](https://github.com/skycoin/hardware-wallet/issues/47#issuecomment-453903791) have to work as it's works in [libskycoin](https://github.com/skycoin/libskycoin/) project, there is no thing like POW in skycoin, it's an issue.
> 
> For each [skycoin-api](https://github.com/skycoin/hardware-wallet/tree/master/skycoin-api) function I have be able find one equivalent in libskycoin and I can't:
> 
> ```
> 
> // extern GoUint32 SKY_cipher_PubKeyFromSig(cipher__Sig* p0, cipher__SHA256* p1, cipher__PubKey* p2);
> int recover_pubkey_from_signed_message(const char* message, const uint8_t* signature, uint8_t* pubkey);
> 
> void ecdsa_get_public_key33(const ecdsa_curve *curve, const uint8_t *priv_key, uint8_t *pub_key);
> // const curve_info *get_curve_by_name(const char *curve_name);
> // int hdnode_from_seed(const uint8_t *seed, int seed_len, const char* curve, HDNode *out);
> // void hdnode_fill_public_key(HDNode *node);
> void create_node(const char* seed_str, HNode* node);
> void uncompress_mcoords(const ecdsa_curve *curve, uint8_t odd, const bignum256 *x, bignum256 *y);
> int mecdsa_validate_pubkey(const ecdsa_curve *curve, const curve_point *pub);
> void mpoint_multiply(const ecdsa_curve *curve, const bignum256 *k, const curve_point *p, curve_point *res);
> void mscalar_multiply(const ecdsa_curve *curve, const bignum256 *k, curve_point *res);
> void mpoint_set_infinity(curve_point *p);
> int mpoint_is_infinity(const curve_point *p);
> void mpoint_add(const ecdsa_curve *curve, const curve_point *cp1, curve_point *cp2);
> // extern GoUint32 SKY_handle_copy(Handle p0, Handle* p1);
> void mpoint_copy(const curve_point *cp1, curve_point *cp2);
> int mpoint_is_equal(const curve_point *p, const curve_point *q);
> int mpoint_is_negative_of(const curve_point *p, const curve_point *q);
> // extern GoUint32 SKY_cipher_DoubleSHA256(GoSlice p0, cipher__SHA256* p1);
> void mpoint_double(const ecdsa_curve *curve, curve_point *cp);
> void mcurve_to_jacobian(const curve_point *p, jacobian_curve_point *jp, const bignum256 *prime);
> void mjacobian_to_curve(const jacobian_curve_point *jp, curve_point *p, const bignum256 *prime);
> void mpoint_jacobian_add(const curve_point *p1, jacobian_curve_point *p2, const ecdsa_curve *curve);
> void mconditional_negate(uint32_t cond, bignum256 *a, const bignum256 *prime);
> void mpoint_jacobian_double(jacobian_curve_point *p, const ecdsa_curve *curve);
> 
> // extern GoUint32 SKY_cipher_ECDH(cipher__PubKey* p0, cipher__SecKey* p1, coin__UxArray* p2);
> void ecdh(const uint8_t* secret_key, const uint8_t* remote_public_key, uint8_t* ecdh_key /*should be size SHA256_DIGEST_LENGTH*/);
> void ecdh_shared_secret(const uint8_t* secret_key, const uint8_t* remote_public_key, uint8_t* shared_secret /*should be size SHA256_DIGEST_LENGTH*/);
> // extern GoUint32 SKY_cipher_SHA256_Set(cipher__SHA256* p0, GoSlice p1);
> void secp256k1Hash(const uint8_t* seed, const size_t seed_length, uint8_t* secp256k1Hash_digest);
> void generate_deterministic_key_pair_iterator(const uint8_t* seed, const size_t seed_length, uint8_t* nextSeed, uint8_t* seckey, uint8_t* pubkey);
> // extern GoUint32 SKY_cipher_SumSHA256(GoSlice p0, cipher__SHA256* p1);
> void compute_sha256sum(const uint8_t *seed, uint8_t* digest /*size SHA256_DIGEST_LENGTH*/, size_t seed_lenght);
> // extern GoUint32 SKY_cipher_PubKeyFromSecKey(cipher__SecKey* p0, cipher__PubKey* p1);
> void generate_pubkey_from_seckey(const uint8_t* seckey, uint8_t* pubkey);
> // extern GoUint32 SKY_cipher_GenerateDeterministicKeyPair(GoSlice p0, cipher__PubKey* p1, cipher__SecKey* p2);
> void generate_deterministic_key_pair(const uint8_t* seed, const size_t seed_length, uint8_t* seckey, uint8_t* pubkey);
> // extern GoUint32 SKY_cipher_AddressFromPubKey(cipher__PubKey* p0, cipher__Address* p1);
> void generate_base58_address_from_pubkey(const uint8_t* pubkey, char* address, size_t *size_address);
> // extern void SKY_cipher_BitcoinAddressFromPubKey(cipher__PubKey* p0, cipher__BitcoinAddress* p1);
> void generate_bitcoin_address_from_pubkey(const uint8_t* pubkey, char* address, size_t *size_address);
> // extern GoUint32 SKY_cipher_BitcoinAddressFromSecKey(cipher__SecKey* p0, cipher__BitcoinAddress* p1);
> void generate_bitcoin_private_address_from_seckey(const uint8_t* pubkey, char* address, size_t *size_address);
> int ecdsa_skycoin_sign(const uint32_t nonce_value, const uint8_t *priv_key, const uint8_t *digest, uint8_t *sig);
> // extern GoUint32 SKY_base58_String2Hex(GoString p0, coin__UxArray* p1);
> void tohex(char * str, const uint8_t* buffer, int bufferLength);
> // extern GoUint32 SKY_cipher_SecKeyFromHex(GoString p0, cipher__SecKey* p1);
> void writebuf_fromhexstr(const char *str, uint8_t* buf);
> ```
> 
> A valuable feature could be the possibility of running [(a subset of) these tests](https://github.com/skycoin/libskycoin/tree/develop/lib/cgo/tests) against [skycoin-api](https://github.com/skycoin/hardware-wallet/tree/master/skycoin-api), so I am trying to figure out how to make this two APIs to match but it result very difficult for me, maybe I am wrong at all or I am missing some little details about the project but I would be very grateful that someone could help me understand.

will be continue in https://github.com/skycoin/hardware-wallet/issues/84

closed 2019-07-09 by gz-c