skycoin/hardware-wallet #54: Improve the include paths

stdevAlDen, 2019-01-18:
- [ ] Use `-I` flag for the project headers.

- [ ] Specify include paths in such a way that you have to include the file with a full path starting from the root source in the project with out using unix shortcuts like `.` or `../`.
- [ ] Use `-isystem` flag for headers in external (3rdparty dependendcy) project. This is very important for some static analizer tools and the compiler itselft.

related to #46

YuraYelisieiev, 2019-04-27:
Check pls that this issue is still valid.

stdevAlDen, 2019-04-29:
> Check pls that this issue is still valid.

@YuraYelisieiev yes, still valid

YuraYelisieiev, 2019-05-01:
Can`t spot a place where I can improve them. Could you help?

stdevAlDen, 2019-05-02:
> Can`t spot a place where I can improve them. Could you help?

@YuraYelisieiev 
Look at this:
https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/Makefile.include#L88-L94

This allow you to to make an include like this:
https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/fsm_impl.c#L31

You can read this related info:
https://google.github.io/styleguide/cppguide.html#Names_and_Order_of_Includes

It's important to include any fix in a linter.

stdevAlDen, 2019-05-02:
> @YuraYelisieiev
> Look at this:
> https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/Makefile.include#L88-L94
>
> This allow you to to make an include like this: [...]

And please, get noticed that this it's not for this milestone

YuraYelisieiev, 2019-05-03:
@stdevAlDen Seems like the first item in the checklist is already done. 
All the headers are included in the way you mentioned.

YuraYelisieiev, 2019-05-03:
What do mean include in a linter?

YuraYelisieiev, 2019-05-03:
@stdevAlDen  Also what`s happening in this line. https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/Makefile#L42 Prolly missing something.

stdevAlDen, 2019-05-03:
> What do mean include in a linter?

@YuraYelisieiev pls mention my user on comments, this improve the notification, so your question:

it depends on [this](https://github.com/skycoin/hardware-wallet/issues/69) but it's stopped for now, I  means If you make any fix this should be enforced by the linter to avoid fall down again.

PD: please note that we already have clang selected but this pr it's locked for now, and clang format have an option([`SortIncludes`](https://clang.llvm.org/docs/ClangFormatStyleOptions.html)) that can be used for some of these points.

stdevAlDen, 2019-05-03:
> @stdevAlDen Also what`s happening in this line. https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/Makefile#L42 Prolly missing something.

@YuraYelisieiev sorry, I not understand your question, in any case you can read a bit more about `-I`, from `man gcc`:
"Options for Directory Search
       These options specify directories to search for header files, for libraries and for parts of the compiler:

       -I dir
       -iquote dir
       -isystem dir
       -idirafter dir
           Add the directory dir to the list of directories to be searched for header files during preprocessing.  If dir begins with =, then the = is replaced by the
           sysroot prefix; see --sysroot and -isysroot.

           Directories specified with -iquote apply only to the quote form of the directive, "#include "file"".  Directories specified with -I, -isystem, or -idirafter
           apply to lookup for both the "#include "file"" and "#include <file>" directives.

           You can specify any number or combination of these options on the command line to search for header files in several directories.  The lookup order is as
           follows:

           1.  For the quote form of the include directive, the directory of the current file is searched first.

           2.  For the quote form of the include directive, the directories specified by -iquote options are searched in left-to-right order, as they appear on the
               command line.
           3.  Directories specified with -I options are scanned in left-to-right order.

           4.  Directories specified with -isystem options are scanned in left-to-right order.

           5.  Standard system directories are scanned.

           6.  Directories specified with -idirafter options are scanned in left-to-right order.

           You can use -I to override a system header file, substituting your own version, since these directories are searched before the standard system header file
           directories.  However, you should not use this option to add directories that contain vendor-supplied system header files; use -isystem for that.

           The -isystem and -idirafter options also mark the directory as a system directory, so that it gets the same special treatment that is applied to the
           standard system directories.

           If a standard system include directory, or a directory specified with -isystem, is also specified with -I, the -I option is ignored.  The directory is still
           searched but as a system directory at its normal position in the system include chain.  This is to ensure that GCC's procedure to fix buggy system headers
           and the ordering for the "#include_next" directive are not inadvertently changed.  If you really need to change the search order for system directories, use
           the -nostdinc and/or -isystem options.
"

YuraYelisieiev, 2019-05-03:
@stdevAlDen Yeah, I know the what it is used for. But [here](https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/Makefile#L42 ) adding INC variable doesn`t work(I can`t include headers from firmware folder in the files which are compiled with this make file). Maybe this 'move' does something which I don`t know.

stdevAlDen, 2019-05-03:
> @stdevAlDen Yeah, I know the what it is used for. But [here](https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/Makefile#L42) adding INC variable doesn't work(I cant include [...]

@YuraYelisieiev Ok, I'm not sure about this but in any case this is what the issue is for, if you whant to ibclude a file from firmware you need to be forced to write `#include "tiny-firware/firmware/file.h"`, If you are able to write `#include "file.h"` without a build error this is an issue.

YuraYelisieiev, 2019-05-03:
@stdevAlDen  Not it seems for me that the first and the second task are the opposite, maybe I didn`t understand the first one correct. Could you specify, please?

stdevAlDen, 2019-05-03:
> @stdevAlDen Not it seems for me that the first and the second task are the opposite, maybe I didn`t understand the first one correct. Could you specify, please?

@YuraYelisieiev Yeah, the 3 points are valid.

YuraYelisieiev, 2019-05-03:
@stdevAlDen If you want to include the file from the directories like firmware, we don`t need to use -I flag for project headers directories because then we include with the header name only. So if the issue is to specify the full path for every project header, I need to remove the -I flag's and specify the full path. But this contradicts the first task.

stdevAlDen, 2019-05-03:
> @stdevAlDen If you want to include the file from the directories like firmware, we don`t need to use -I flag for project headers directories because then we include with the header name only. So if [...]

@YuraYelisieiev Let me try to explain in a different way: For the whole task you need to remove ol the search path for header files currently set up explicitly to the compiler, but keeping a `-I` for the root project dir, a `-isystem` for the vendors folder and refactor the source code (and maybe other details I'm missing now).

What do you think? are this equivalent to the 3 original points o you recommend me to change the description?

YuraYelisieiev, 2019-05-03:
@stdevAlDen Thanks, now I understand it completely. There was just a lack of the issue description. That's why there was a misunderstanding.

stdevAlDen, 2019-05-03:
> @stdevAlDen Thanks, now I understand it completely. There was just a lack of the issue description. That's why there was a misunderstanding.

@YuraYelisieiev Oh I'm sorry, but I feel happy that you finally understand and you can start with the issue.

YuraYelisieiev, 2019-05-03:
@stdevAlDen Just to make sure I am doing right. If I am in the `tiny-firmware/firmware` folder. To include a header from the previous directory I write `#include tiny-firmware/some.h` and so on. Right ?

YuraYelisieiev, 2019-05-10:
#227 @stdevAlDen

jdknives, 2019-09-20:
Closed by #227

closed 2019-09-20 by jdknives