skycoin/hardware-wallet #57: List make targets

olemis, 2019-01-19:
**Is your feature request related to a problem? Please describe.**
There is no easy way to list available `make` targets

**Describe the solution you'd like**
Implement `make help`

**Possible implementation**
Should be similar to `skycoin/skycoin` 's

mpsido, 2019-01-19:
In my environment I open a console (terminator console), I go to my project folder and type `make` then tab tab. I have this:
```
mpsido@mpsido:~/my_projects/skycoin/hardware-wallet$ make 
bootloader                 build-deps                 clean-lib                  firmware                   full-firmware              run-emulator               st-skycoin
bootloader-mem-protect     clean                      emulator                   firmware-deps              full-firmware-mem-protect  sign                       tiny-firmware/

```

olemis, 2019-01-19:
Sure , the changes are aimed at achieving something like 

```
$ make help
clean-lib                      Delete all files generated by tiny-firmware library dependencies
clean                          Delete all files generated by build
build-deps                     Build common dependencies (protob, nenopb)
firmware-deps                  Build firmware dependencies
bootloader                     Build bootloader
bootloader-mem-protect         Build bootloader
firmware                       Build wallet firmware
build-libc                     Build the Skycoin cipher library for firmware
sign                           Sign wallet firmware
full-firmware-mem-protect      Build full firmware (no mem protect)
full-firmware                  Build full firmware (no mem protect)
emulator                       Build emulator
run-emulator                   Run wallet emulator
st-flash                       Deploy (flash) firmware on physical wallet
```

I think I'd also take the time to revise and learn from a few targets while doing so .

olemis, 2019-01-19:
Indeed , now I notice some changes needed

PR #58 (2019-01-30): fixes #57 fixes #59 - Implement make help. OSX Travis builds 
Fixes #57 
Fixes #59

Changes:
- Add `make help` target in for `skycoin-api` and top level `Makefile`'s
- Document `make` targets
- Absolute paths in `skycoin-api` `Makefile` .
- Support for `macos`

Needs to be mentioned in CHANGELOG:
yes
closed 2019-01-30 by olemis