skycoin/hardware-wallet #61: Use libbsd in GNU/Linux for emulator

stdevAlDen, 2019-01-22:
Remove [`strl.h`](https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/emulator/strl.h) and [`strl.c`](https://github.com/skycoin/hardware-wallet/blob/master/tiny-firmware/emulator/strl.c) and use `libbsd-dev` instead for the emulator in GNU/Linux.

Remember to link against bsd lib [on linux](https://gist.github.com/sighingnow/deee806603ec9274fd47#file-makefile-L15)

stdevHan, 2019-04-28:
@stdevAlDen I would like to know what would be good to discard those files from the compilation in `GNU/Linux`, if at the end you have to have them for `OSX`?

stdevAlDen, 2019-04-29:
@stdevHan the `strlcpy` functionality it's not standard, it's a `BSD` extension, as osx belongs to the bsd family this feature is available, in GNU/Linux you need to have a compatible alternative, so as this function is currently used in the project it's required, what I recommend is move it out from the source code tree that we maintain and use a 3rd party library(`libbsd-dev`) if you think that this library is is equivalent to the current code what I recommend is just to move it to vendor folder, it's just my opinion, let me know what do you think.

stdevAlDen, 2019-04-29:
> @stdevHan the `strlcpy` functionality it's not standard, it's a `BSD` extension, as osx belongs to the bsd family this feature is available, in GNU/Linux you need to have a compatible alternative, [...]

@stdevHan In any case this no look relevant for this milestone, is open for discussion.