skycoin/hardware-wallet #64: Firmware versioning

olemis, 2019-01-25:
A documented process is needed for managing the version number during development and for releases .

jdknives, 2019-01-29:
We should be using semantic versioning for the firmware releases generally. Check this document for reference: https://semver.org/

olemis, 2019-01-30:
In other projects we've a versioning model that resembles Skycoin . For instance PySkycoin's major and minor version numbers are exactly the same as Skycoin node's at release time . There is a source code dependency in that case , higher coupling than in here (desktop wallet => hw wallet)

olemis, 2019-02-07:
Semantic version scheme for consecutive release builds of `tiny-firmware` is ok . Nonetheless for combined firmware I'd rather prefer an scheme that considers.

- **Bootloader version**: bootloader version identifier
- **Firmware version**: identifier for tiny-firmware version number
- **SoC chip** MCU model identifier.
- **Country Exit Code**: to cope with i18n and locale specific features

olemis, 2019-03-03:
Once skycoin/hardware-wallet-protob#4 will be merged , firmware release implies tagging `skycoin/hardware-wallet-protob` to match versions .

olemis, 2019-04-23:
@gz-c combined firmware versioning scheme should consider locale and regional settings . Will we ever have more than 32 officially supported target languages ? Officially supported means for mainstream milestone releases .

olemis, 2019-04-24:
> Will we ever have more than 32 officially supported target languages ?

We shall focus on [supported BIP39 word lists](https://github.com/bitcoin/bips/blob/master/bip-0039/bip-0039-wordlists.md).

olemis, 2019-05-03:
fixed in 5860c3e

PR #75 (2019-02-07): fixes #50 fixes #64 - Release process and Makefile targets
Fixes #50 

Changes:
- Add `bootloader-clean`, `bootloader-release`, `bootloader-release-mem-protect` make targets
- Add `firmware-clean`, `firmware-release` make targets
- Add `combined-release`, `combined-release-mem-protect` make targets

Does this change need to mentioned in CHANGELOG.md?
yes

Requires testing
no
commit e418401d (2019-02-07): Merge pull request #75 from simelo/olemis_t50_makefiles

fixes #50 fixes #64 - Release process and Makefile targets
PR #183 (2019-05-03): Create a firmware version deployment usable from makefile fixes #50
Fixes #50
closes #64 
closes #155 
closes #22 

Changes:
- Remove releases folder
- Remove all the protob build related commands, use `build-c` from submodule only.
- Add some ci related bash scrips.
- Call `make release` when the travis build has been triggered by a `git tag`.
- Add deploy satge for travis.
- Get firmware version from `git`, use `tiny-firmware/VERSION` if it's not possible.
- Refactor in makefile targets, put a `release-` prefix for all release related targets.
- Update Readme(ol content from `releases/REAMDE [... cut]
closed 2019-05-03 by olemis