skycoin/hardware-wallet #77: Request confirmation to sign generic messages

Senyoret1, 2019-02-10:
**Feature description**
The hardware wallet has a function that allows to sign generic messages. Previously the function was used to sign transactions and the user was not requested for confirmation due to several UX reasons, but now there is a function that allows to sign complete transactions, so the function that signs generic messages will not be used in the integration with the desktop wallet any more (the change is in https://github.com/skycoin/skycoin/pull/2156). Due to this change, now it would be good to modify the function that signs generic messages to make it ask the user for confirmation.

**Is your feature request related to a problem? Please describe.**
Requesting confirmation would increase security, since it would not be possible for malicious software to sign messages with the conneted hardware wallet without the user's approval.

**Describe the solution you'd like**
While asking for confirmation, the hardware wallet screen may show the message that will be signed. The operation could be similar to the function that allows signing transactions.

**Describe alternatives you've considered**
Another alternative could be to delete the function that allows to sign generic messages, since at the moment it would not be used. However, it may be useful in the future.

jdknives, 2019-03-09:
Can this simply be deactivated for v1.0.0?

Senyoret1, 2019-03-09:
I guess so, I think the function is not used for anything, at least for now.

jdknives, 2019-03-09:
OK. Then this should be done.

olemis, 2019-04-04:
> Requesting confirmation would increase security, since it would not be possible for malicious software to sign messages with the conneted hardware wallet without the user's approval.

Message signing is protected by PIN .

stdevStark, 2019-04-08:
@olemis why buttons can not be pushed in 2be2842 ?

olemis, 2019-04-08:
@stdevStark I do not understand your question ...

olemis, 2019-04-11:
> @olemis why buttons can not be pushed in [2be2842](https://github.com/skycoin/hardware-wallet/commit/2be2842d6ad185aeca226fb486daa2e843bcc1dd) ?

i am not sure

stdevStark, 2019-05-04:
> i am not sure

see skycoin/hardware-wallet-js#114

olemis, 2019-05-05:
fixed in 160614a

PR #213 (2019-05-05): fixes #77 - Ask confirmation before Skycoin sign message
Fixes #77

Changes:
- Ask confirmation before Skycoin sign message

Does this change need to mentioned in CHANGELOG.md?
yes

Requires changes in protobuff specs?
no

Requires testing
no

the change is in `fsm.c` not implementation . Tested with hw-js suite in Travis .
commit 160614a3 (2019-05-05): Merge pull request #213 from simelo/stdevStark_t77_signmsg_pinprotect

fixes #77 - Ask confirmation before Skycoin sign message
closed 2019-05-05 by olemis