skycoin/hardware-wallet #78: The hardware wallet shows only up to 2 decimals

Senyoret1, 2019-02-10:
**Describe the bug**
When the function for signing transactions is used (added in https://github.com/skycoin/hardware-wallet/pull/62 ), the hardware wallet screen shows the details of the outputs. The problem is that the coins are always shown with 2 decimals, being that this value could have up to 6 decimals.

**Environment information:**

- OS: any
- Platform: any
- Firmware Version: tested with the firmware updated until https://github.com/skycoin/hardware-wallet/commit/e418401d19bc03ce4a366550e479c7bf487632cf

**Steps to Reproduce**
The function can be executed with the JS library and with the cli.

**Actual behavior**
Explained at the beginning.

**Expected behavior**
The hardware wallet should be able to show more decimals. Additionally, it would be good to hide the unnecessary decimals (like showing “1” instead of “1.00”).

olemis, 2019-02-12:
@gz-c should `hardware-wallet` respect `MaxDropletPrecision` (currently `3` for Skycoin) fiber parameter ?

gz-c, 2019-02-12:
no that it should not validate soft parameters. i'm not sure it needs to validate the transaction at all, the client can do it

olemis, 2019-02-12:
My question was just about using that value for displaying numbers in hardware wallet display , not txn validations. In any case , how many decimal digits should be displayed?

Senyoret1, 2019-02-12:
I think that it is not necessary to respect that value. At this time the function on the hardware wallet receives the amount of coins in droplets, so we could say that there is always the possibility of receiving up to 6 decimals. The function should simply show the minimum number of decimals possible. For example:

- If the function receives `2010000` it shoud display `2.01` (the result of `2010000/1000000`).
- If the function receives `2000000` it shoud display `2`.
- If the function receives `2000001` it shoud display `2.000001`.

The client should take care of not sending more decimals than allowed.

olemis, 2019-02-16:
fixed by #93 in e96ea83 .

PR #93 (2019-02-16): [firmware] fixes #78 #91 - Coin fractions in Skywallet display
Fixes #78
Fixes #91 

Changes:
- Change byte size of coins and hours `uint32_t` => `uint64_t`
- Function to render integer coin values as requested in #78 (no conversion to float)
- Include path properly set to run `make test` on osx and linux platforms

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes

Tested with check. Test code refactored to have a `main()` function and a separate mechanism to add test cases .

Related to skycoin/hardware-wallet-js#56 and skycoin/hardware-wallet-go#35
commit e96ea830 (2019-02-16): Merge pull request #93 from simelo/olemis_t78_coin_fraction_display

[firmware] fixes #78 fixes #91 - Coin fractions in Skywallet display
closed 2019-02-16 by olemis