skycoin/hardware-wallet #81: Integer error codes

olemis, 2019-02-12:
**Feature description**

Describe the feature

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the problem here.

**Possible implementation**
Preliminary ideas to get this done.
If you have all details in mind then provide a checklist

- [ ] Start with ...
- [ ] When done do ...
- [ ] Finally ...

jdknives, 2019-03-09:
This issue does not specify a whole lot. Either it needs to be closed or the problem needs to be described. @olemis

olemis, 2019-03-22:
> This issue does not specify a whole lot. Either it needs to be closed or the problem needs to be described. @olemis

Sorry , did not notice once I created the issue .
This is already implemented in `error.h`

closed 2019-03-22 by olemis