skycoin/hardware-wallet #84: Tests for hw wallet skycoin-api in compliance to Skycoin cipher API

olemis, 2019-02-13:
**Feature description**

Under all circumstances `skycoin-api` must be 100% compatible with `skycoin/skycoin`

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Check for a match between `skycoin-api` input arguments and corresponding results by using values in `skycoin/skycoin` golden files.

**Additional context**
Should be limited to the subset of the Skycoin API needed in hw wallet. Maybe related to skycoin/libskycoin#7 .

**Possible implementation**

TODO: Detailed steps

stdevAlDen, 2019-05-02:
Possible related comment https://github.com/skycoin/hardware-wallet/issues/47#issuecomment-457768216

olemis, 2019-07-17:
not v1 feature