skycoin/hardware-wallet #88: Seed Generation / Entropy Source Questions

gz-c, 2019-02-14:
We need to find the answers to these questions, and document them:

0. Why is it using `rand()` in this file? https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/fsm_impl.c
0. How does `rand()` in the firmware generate random numbers?
0. How are seeds generated by the hardware wallet?

https://www.reddit.com/r/Bitcoin/comments/2d36lb/is_the_trezor_seed_entropy_truly_and_provably/

The link above implies that Trezor uses a source of entropy from the host machine and from the hardware wallet itself

0. Does our hw wallet have an entropy source?
0. Does our hw wallet get entropy from the host machine?
0. How does our hardware wallet get entropy?
0. How does our hardware wallet ensure we have enough entropy to generate a seed?

gz-c, 2019-02-14:
There is a RNG test tool here:
https://github.com/trezor/rng-test

stdevAlDen, 2019-02-19:
WIP
> We need to find the answers to these questions, and document them:
> 
>     1. Why is it using `rand()` in this file? https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/fsm_impl.c
Look like a critical bug.
> 
>     2. How does `rand()` in the firmware generate random numbers?
It's a pseudo random number generator.
> 
>     3. How are seeds generated by the hardware wallet?
> 
> 
> https://www.reddit.com/r/Bitcoin/comments/2d36lb/is_the_trezor_seed_entropy_truly_and_provably/
> 
> The link above implies that Trezor uses a source of entropy from the host machine and from the hardware wallet itself
> 
>     1. Does our hw wallet have an entropy source?
Yes
> 
>     2. Does our hw wallet get entropy from the host machine?
Not yet, but it should to get entropy from the host machine **in addition to the one in the physical device**.
> 
>     3. How does our hardware wallet get entropy?
[Is based on an analog circuit. This circuit generates a continuous analog noise that will be 
used on the RNG processing in order to produce a 32-bit random number.](https://www.st.com/content/ccc/resource/technical/document/application_note/4a/6a/82/05/8e/9e/4e/94/DM00073853.pdf/files/DM00073853.pdf/jcr:content/translations/en.DM00073853.pdf)
> 
>     4. How does our hardware wallet ensure we have enough entropy to generate a seed?

stdevAlDen, 2019-02-22:
@gz-c how long should be the external entropy buffer?

gz-c, 2019-02-22:
@stdevAlDen Refer to trezor's source:

https://github.com/trezor/trezor-mcu/blob/master/firmware/reset.c

There is some use of `int_entropy` and `ext_entropy` here - following the path should lead you to the nature of `ext_entropy` (external entropy)

stdevAlDen, 2019-02-22:
> @stdevAlDen Refer to trezor's source:
> 
> https://github.com/trezor/trezor-mcu/blob/master/firmware/reset.c
> 
> There is some use of `int_entropy` and `ext_entropy` here - following the path should lead you to the nature of `ext_entropy` (external entropy)

Thank you @gz-c I found it [here](https://github.com/trezor/trezor-common/blob/0735c7d6f524b4c5108d201c789612aad7ce7920/protob/messages-management.proto#L215), I thing that may be great if after the hw-device recieves this buffer of external entropy [measure the entropy level](https://rosettacode.org/wiki/Entropy) and return an error if not fit a "minimum" value. What value you recommend for this?, for example some executions get values around 4.4... produced from my Linux laptop, a end user msg could be:
"No enough entropy in your device, make a crazy use of your keyboard"

@olemis & @gz-c thank you in advance.

gz-c, 2019-02-22:
Does trezor measure the entropy level? I'm not sure what the minimum value should be; you can sample 32bytes of random data from a linux system and measure the entropy of that to get an idea of what it should be

stdevAlDen, 2019-02-22:
> Does trezor measure the entropy level? I'm not sure what the minimum value should be; you can sample 32bytes of random data from a linux system and measure the entropy of that to get an idea of what it should be

I no saw this in trezor, your approach sounds ok.

stdevAlDen, 2019-02-25:
This weekend I was reading about `rng` validation and saw that in general there are two approaches:
- Create a firmware with that specific purpose, generate the data and then validate
- Add a functionality in the production firmware to read generated numbers, this functionality can be compiled based in a condition flag, for example `ENABLE_VALIDATION_TESTS`

The first one look like a post hardware desig/implementation validation (currently done by st but you can confirm). The second one look like a hardware + firmware validation.
Witch one do you recommend?

olemis, 2019-02-26:
After a bit more of research I'll start replying some questions ...

> 
> > We need to find the answers to these questions, and document them:
> > ```
> > 1. Why is it using `rand()` in this file? https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/fsm_impl.c
> > ```
> 
> Look like a critical bug.
> 

Agreed .

olemis, 2019-02-26:
> > ```
> > 2. How does `rand()` in the firmware generate random numbers?
> > ```
> 
> It's a pseudo random number generator.
> 
> > 1. Does our hw wallet have an entropy source?
> > ```
> 
> Yes
> 

On chip entropy generation, relies on hardware but happens in software .

> [Is based on an analog circuit. This circuit generates a continuous analog noise that will be used on the RNG processing in order to produce a 32-bit random [...]

correct 

Beyond this all ECC primitives implemented in software .

stdevAlDen, 2019-02-28:
> We need to find the answers to these questions, and document them:
> 
>     1. Why is it using `rand()` in this file? https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/fsm_impl.c
I recommend here to remove code like this [`ecdsa_skycoin_sign`](https://github.com/skycoin/hardware-wallet/blob/master/skycoin-api/skycoin_crypto.c#L178) from the project, as this should be a standard implementation could be possible use some other, for an OS supported platform for example `openssl` may be used, for this case what I recommend is using a stm32 [implementation](https://my.st.com/content/my_st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-standard-peripheral-library-expansion/stm32-cryp-lib.license=1551292112296.product=STM32-CRYP-LIB.version=2.0.6.html) if you thing that this could be ok, pls let me know and I can make an in deep look about the possibility of use `STM32-CRYP-LIB` @olemis & @gz-c 

PD: [this line](https://github.com/skycoin/hardware-wallet/blob/master/skycoin-api/skycoin_crypto.c#L191)(among others) look very interesting/strange for me...
> 
>     2. How does `rand()` in the firmware generate random numbers?
> 
>     3. How are seeds generated by the hardware wallet?
> 
> 
> https://www.reddit.com/r/Bitcoin/comments/2d36lb/is_the_trezor_seed_entropy_truly_and_provably/
> 
> The link above implies that Trezor uses a source of entropy from the host machine and from the hardware wallet itself
> 
>     1. Does our hw wallet have an entropy source?
> 
>     2. Does our hw wallet get entropy from the host machine?
> 
>     3. How does our hardware wallet get entropy?
> 
>     4. How does our hardware wallet ensure we have enough entropy to generate a seed?

gz-c, 2019-02-28:
1. Do not change the crypto. Do not use openssl. Openssl was removed as a dependency from bitcoin and replaced with these C libraries

stdevAlDen, 2019-02-28:
>     1. Do not change the crypto. Do not use openssl. Openssl was removed as a dependency from bitcoin and replaced with these C libraries

If I can understand ok the recommended approach should be ([in this line](https://github.com/skycoin/hardware-wallet/blob/230a1d5b76fb210506adc257b619bdd6201e5135/tiny-firmware/firmware/fsm_impl.c#L84) for example) pass as first parameter a true random number?

gz-c, 2019-02-28:
That looks like a nonce; refer to trezor's bitcoin code for a similar example

gz-c, 2019-02-28:
https://www.joachim-breitner.de/blog/749-Nonce_sense_paper_online

gz-c, 2019-02-28:
A CSPRNG number is required; if the nonce is used twice, or the nonces are predictable, the private key will be leaked.  Bitcoin uses deterministic signatures now (2016), removing the need for a random nonce and the risk of it not being true random.  Skycoin still uses nonce'd signatures

olemis, 2019-02-28:
@stdevAlDen I am already looking into this . I started to download library but could not extract the resulting file . it was corrupted

stdevAlDen, 2019-02-28:
> @stdevAlDen I am already looking into this . I started to download library but could not extract the resulting file . it was corrupted

@olemis What library?

olemis, 2019-02-28:
@stdevAlDen STM32-CRYP-LIB

stdevAlDen, 2019-02-28:
> @stdevAlDen STM32-CRYP-LIB

I was not able to find any security certification/validation from the papers, for now we should keep this implementation as is, [if I not misunderstand](https://github.com/skycoin/hardware-wallet/issues/88#issuecomment-468141574), give me some minutes, I'm going to put here some interesting info in some minutes.

stdevAlDen, 2019-02-28:
Biased Nonce Sense: Lattice Attacks against Weak ECDSA Signatures in Cryptocurrencies
===============

Joachim Breitner (DFINITY Foundation, Zug, joachim@dfinity.org) and Nadia Heninger (University of California, San Diego, nadiah@cs.ucsd.edu):
------------------------
- "**Abstract.** In this paper, we compute hundreds of Bitcoin private keys and  dozens  of  Ethereum,  Ripple,  SSH,  and  HTTPS  private  keys  by carrying out cryptanalytic attacks against digital signatures contained in public blockchains and Internet-wide scans. The ECDSA signature algorithm requires the generation of a per-message secret nonce. This nonce must be generated perfectly uniformly, or else an attacker can exploit the nonce biases to compute the long-term signing key."
- "We first use the hidden number problem algorithm to discover the long-term ECDSA signature key when used with nonces that are shorter than expected, and found keys used with nonces with lengths 64 bits, 110 bits, 128 bits, and 160 bits. We extend this technique to discover nonces with shared prefixes and suffixes, and found keys used with signature nonces that shared prefixes and varied in their 64 least significant bits, as well as keys used with signature nonces that shared suffixes and varied in their 128 and 224 most significant bits."
- "They  traced  the  problems  primarily  back  to implementation vulnerabilities in random number generation **on low-resource devices**."
- "Countermeasures. All  of  the  attacks  we  discuss  in  this  paper  can  be  prevented by using deterministic ECDSA nonce generation [29], which is already implemented in the default Bitcoin and Ethereum libraries."
  - @gz-c :point_up_2: maybe relate to you comment "Bitcoin uses deterministic signatures now (2016), removing the need for a random nonce and the risk of it not being true random"

The above are just some lines from [the paper](https://eprint.iacr.org/2019/023.pdf)

olemis, 2019-03-13:
@gz-c I'm considering this closed after merging `stdevAlDen_t88_seed_generation_entropy_source_questions` in 1de337d .
Feel free to reopen it or create a follow up issue if something is missing (answers to questions in FAQ)

PR #106 (2019-03-13): External entropy generation and merge - fixes #88
Fixes #88

Changes:
- Add a "Frequently Asked Question" file.
- Remove the `objconv` instalation for osx, this is not required any more.
- Use a `EntropyRequest` / `EntropyAck` schema to get more entropy if the internal one is not enough.
- In `msgGenerateMnemonicImpl` use 12 as word count only if not specify any value.
- Use a true random number as nonce value in `ecdsa_skycoin_sign`.

It's required for some security/crypto expert to check out this PR, specifically the [engout_entropy](https://github.com/simelo/skycoin-hardware-wallet/commit/1 [... cut]
closed 2019-03-13 by olemis