skycoin/hardware-wallet #98: GetVersion does not return the firmware version

therealssj, 2019-02-19:
**Describe the bug**
The `GetVersion` message does not return the firmware version instead it returns `success`(?)

**Actual behavior**
Returns success

https://github.com/skycoin/hardware-wallet/blob/39229d610a4f8d68d9f4f30921fd9667ea52b4b7/tiny-firmware/protob/messages.proto#L307-L312

**Expected behavior**
Return firmware version

stdevAlDen, 2019-02-19:
@therealssj & @olemis 
I recommend remove this message and use [`Featres`](https://github.com/skycoin/hardware-wallet/blob/39229d610a4f8d68d9f4f30921fd9667ea52b4b7/tiny-firmware/protob/messages.proto#L100-L102) instead, in addition write a unit tests for this, what do you think?

olemis, 2019-02-19:
@therealssj indeed these are the right values to consider

https://github.com/skycoin/hardware-wallet/blob/39229d610a4f8d68d9f4f30921fd9667ea52b4b7/tiny-firmware/protob/messages.proto#L84-L86

They contain the version numbers of the feature in use (either bootloader or firmware). The ones mentioned above by @stdevAlDen are to report firmware version number when in bootloader mode.

olemis, 2019-02-19:
`GetVersion` message was added in 7fe699d9 and has not changed much since then . Does not seem to be at all useful. If we all agree I'm keen of removing it.

This is low priority though . Not a reason for blocking first release .

olemis, 2019-02-19:
On the other hand @therealssj , success message has a text field . That's enough for returning firmware version back to the connected peer. That's the reason why there is no specific message to reply to `GetVersion` requests.

stdevStark, 2019-02-20:
@olemis fixed in f351c5f but issue not closed

PR #101 (2019-02-20): fixes #98 -  Remove GetVersion message
Fixes #98 

Changes:
- Remove `fsm_msgGetVersion` in `fsm.c`
- Remove related `protobuf` message definitions

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
no
commit aa34aaf7 (2019-02-20): Merge pull request #101 from simelo/stdevStark_t98_rm_getversion

fixes #98 -  Remove GetVersion message
closed 2019-02-20 by olemis