skycoin/hardware-wallet #103: Fix field size in protobuff message.

stdevAlDen, 2019-02-26:
**Describe the bug**
A difference in symmetric(send/receive some field from two messages) protobuf message.

**Environment information:**

- Any

**Steps to Reproduce**
Check the source code, [this sign](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/protob/messages.options#L25) a message, and [this check](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/protob/messages.options#L23) the message signature.

**Actual behavior**
The messages fields are different.

**Expected behavior**
The messages fields should have the same size.

**Possible implementation**
[SkycoinSignMessage.message](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/protob/messages.options#L25) should have a size of 65.

olemis, 2019-03-05:
@stdevAlDen will this be fixed after integrating `skycoin/hardware-wallet-protob` ?

stdevAlDen, 2019-03-21:
> ed after i

https://github.com/skycoin/hardware-wallet-protob/issues/2

closed 2019-03-21 by stdevAlDen