skycoin/hardware-wallet #104: List security relevant crypto files

jdknives, 2019-02-28:
**Feature description**

This list can be the basis for a final review of the crypto for the Skywallet from Synth.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Simple list of files that make review easier.

**Describe alternatives you've considered**
Alternative is to give him the whole codebase for review, which is unnecessary and wasting time. 

**Additional context**
Add any other context about the problem here.

**Possible implementation**
Preliminary ideas to get this done.
If you have all details in mind then provide a checklist

- [ ] Start with ...
- [ ] When done do ...
- [ ] Finally ...

gz-c, 2019-03-10:
Key topics: 

- Secret key / public key generation
- pubkey to address function
- sha256
- ripemd160
- secp256k1 signatures

stdevAlDen, 2019-04-05:
preliminary list (WIP, next step, put specific links):
==========================

hardware-wallet/tiny-firmware/bootloader/signatures.c
hardware-wallet/tiny-firmware/bootloader/bootloader.c
hardware-wallet/tiny-firmware/firmware/storage.c
hardware-wallet/tiny-firmware/firmware/reset.c
hardware-wallet/tiny-firmware/firmware/recovery.c
hardware-wallet/tiny-firmware/firmware/fsm.c
hardware-wallet/tiny-firmware/firmware/fsm_impl.c
hardware-wallet/tiny-firmware/firmware/entropy.c
hardware-wallet/tiny-firmware/setup.c
hardware-wallet/tiny-firmware/serialno.c
hardware-wallet/tiny-firmware/rng.c
hardware-wallet/tiny-firmware/memory.ld
hardware-wallet/tiny-firmware/memory.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/test_skycoin_crypto.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/skycoin_crypto.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/rand.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/hmac.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/bip32.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/curves.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/sha2.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/secp256k1.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/blake256.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/base58.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/bip39.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/pbkdf2.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/bignum.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/ripemd160.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/ecdsa.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/memzero.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/tools/hasher.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/skycoin_check_signature_tools.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/skycoin_check_signature.c
hardware-wallet/tiny-firmware/vendor/skycoin-crypto/check_digest.c

olemis, 2019-04-09:
@gz-c I'm closing this now , should you need any further actions on this item , please feel free to reopen it .

stdevAlDen, 2019-06-06:
> # Security related files
> ## Crypto core:
> * most files under vendor/skycoin-crypto/tools(or the same, hardware-wallet/skycoin-api/tools) [come from trezor crypto project](https://github.com/trezor/trezor-crypto)
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/rand.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/hmac.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/bip32.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/curves.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/sha2.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/secp256k1.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/blake256.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/base58.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/bip39.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/pbkdf2.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/bignum.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/ripemd160.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/ecdsa.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/memzero.c
>   https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/tools/hasher.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/rng.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/skycoin_crypto.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/skycoin_check_signature_tools.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/skycoin_check_signature.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/check_digest.c
> 
> ## Related to crypto:
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/bootloader/signatures.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/bootloader/bootloader.c
>   
>   * some little details like `int signed_firmware = signatures_ok(hash);` or `__stack_chk_guard = random32();`
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/bootloader/usb.c
>   
>   * some little details like `if (brand_new_firmware || old_was_unsigned || (flags & 0x01) == 0 || SIG_OK != signatures_ok(NULL)) {`
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/setup.c
>   
>   * find this `to be extra careful and heed the STM32F205xx Reference manual, Section 20.3.1`
>     , this `"Each subsequent generated random number has to be compared with the previously generated`
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/serialno.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/memory.c
>   
>   * some thing like `memory_bootloader_hash`
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/firmware/entropy.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/skycoin-api/test_skycoin_crypto.c
> 
> ## General security (can involve some code using crypto too):
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/memory.ld
>   
>   * it's about the defined confidential section
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/firmware/storage.c
>   
>   * multiple little details like `reset_entropy_mix_256`
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/firmware/reset.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/firmware/recovery.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/firmware/fsm.c
> * https://github.com/skycoin/hardware-wallet/tree/develop/tiny-firmware/firmware/fsm_impl.c
>   
>   * https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/test_fsm.c
> 
> ## Update - Thu 06 Jun 2019 02:48:43 PM CDT
@gz-c || @jdknives please let me know if you get notified on this update.
> In the [`TestAddressVerify`](https://github.com/simelo/libskycoin/blob/cbcc513204481127b184fe44da745f2677a39978/lib/cgo/tests/check_cipher.address.common.c#L16-L42) there are a [first](https://github.com/simelo/libskycoin/blob/cbcc513204481127b184fe44da745f2677a39978/lib/cgo/tests/check_cipher.address.common.c#L23) and a [second](https://github.com/simelo/libskycoin/blob/cbcc513204481127b184fe44da745f2677a39978/lib/cgo/tests/check_cipher.address.common.c#L30) call to `SKY_cipher_GenerateKeyPair`. Looking at the [`SKY_cipher_GenerateKeyPair`](https://github.com/simelo/skycoin-hardware-wallet/blob/97c1c82dee752ed00178dd6f19c3161dd36f5a2d/skycoin-api/libskycoin-wrapper.c#L36-L43) implementation (wrapper) looks like (the test was failing) [`srand`](https://github.com/simelo/skycoin-hardware-wallet/blob/97c1c82dee752ed00178dd6f19c3161dd36f5a2d/skycoin-api/libskycoin-wrapper.c#L38) is receiving the same value the two times, this problem is solved in [this commit](https://github.com/simelo/skycoin-hardware-wallet/commit/fe51d79b44d06d7ecd781a3a3ba48732bb899981), I think that is interesting to look at this, and in general to look at the required API changes (wrapper) to make some libskycoin tests work against hardware-wallet API.
> [hardware-wallet/skycoin-api/libskycoin-wrapper.c](https://github.com/simelo/skycoin-hardware-wallet/blob/stdevAlDen_t84_equivalent_skycoin_cipher_tests/skycoin-api/libskycoin-wrapper.c)

jdknives, 2019-06-07:
@stdevAlDen you want someone to look at the changes you have done if I understand correctly?

stdevAlDen, 2019-06-07:
> @stdevAlDen you want someone to look at the changes you have done if I understand correctly?

@jdknives A review it's welcome but it's not what I means with my previous message, let me put you in context. I'm working in [this issue](https://github.com/skycoin/hardware-wallet/issues/84), this is to run the [libskycoin test battery](https://github.com/skycoin/libskycoin/tree/785df27662e57b8ecd78644c7e212b7d2c5d9f43/lib/cgo/tests)(some of they, this does not apply for some tests) against the [hardware-wallet API](https://github.com/skycoin/hardware-wallet/tree/a9e7d9252b578a26c0caaf19c4c230c6c03d8685/skycoin-api), but because the libskycoin and hardware-wallet crypto API are not compatible I need to [adapt](https://en.wikipedia.org/wiki/Adapter_pattern) the hardware-wallet crypto API to be as required by the current libskycoin tests. I means **I have not date changes at all in the hardware-wallet crypto API for this issue**(I'm going to put some exceptions later in the pr), it's just a wrapper, but looking at this wrapper a cryptographic exper can get relevant info instantaneously, for example:
If you look at the [`PubKeyRipemd160`](https://github.com/skycoin/skycoin/blob/develop/src/cipher/address.go#L54-L59) implementation in skycoin you can get noticed instantaneously that this implementation does not exist in hardware-wallet crypto API looking at the [`SKY_cipher_HashRipemd160`](https://github.com/simelo/skycoin-hardware-wallet/blob/347665d822c480b366415ffdb22c6136e0d7c0c0/skycoin-api/libskycoin-wrapper.c#L206-L214) wrapped function. Later looking at how this is used in [`generate_base58_address_from_pubkey`](https://github.com/simelo/skycoin-hardware-wallet/blob/347665d822c480b366415ffdb22c6136e0d7c0c0/skycoin-api/skycoin_crypto.c#L197-L199) maybe we no need get alarmed about ripemd160 API/usage, but in general what I means is that after merged, it's interesting to look at the libskycoin wrapper, and of course this look can found a misuse from my side. Please let me know if you can understand me now.

olemis, 2019-06-20:
audit performed by @gz-c

closed 2019-06-20 by olemis