skycoin/hardware-wallet #105: Validate the device RNG.

stdevAlDen, 2019-02-28:
**Feature description**

Create some `protobuf` messages(maybe compile optionally with a flag) to validate the `rng` in the device.

**Is your feature request related to a problem? Please describe.**
Just to make sure about [the security](https://github.com/skycoin/hardware-wallet/issues/88).

**Describe the solution you'd like**
Be able to run [trezor](https://github.com/trezor/rng-test) and [st](https://www.st.com/content/ccc/resource/technical/document/application_note/4a/6a/82/05/8e/9e/4e/94/DM00073853.pdf/files/DM00073853.pdf/jcr:content/translations/en.DM00073853.pdf) `rng` validations, update the FAQ

**Describe alternatives you've considered**
Make a separate/specific firmware or create a feature for this purpose in this firmware.

**Additional context**
https://github.com/skycoin/hardware-wallet/issues/88

**Possible implementation**
Define a(or some) `protobuf` message, get entropy from the device, save this in a regular pc storage and run statistic tests (like [3.2.2 Second step: NIST statistical test](https://www.st.com/content/ccc/resource/technical/document/application_note/4a/6a/82/05/8e/9e/4e/94/DM00073853.pdf/files/DM00073853.pdf/jcr:content/translations/en.DM00073853.pdf) for example) over this data.

olemis, 2019-03-23:
@stdevAlDen if this is to validate device TRNG is it possible to use some existing code for this ?

stdevAlDen, 2019-03-27:
> @stdevAlDen if this is to validate device TRNG is it possible to use some existing code for this ?

@olemis sure, but I recomment to make (and maintain) a fork of https://github.com/trezor/rng-test and add [this feature](https://github.com/trezor/python-trezor/blob/master/tools/rng_entropy_collector.py) to hardware-wallet-go

PD: I'm having problems to make `python-trezor` and/or `rng-test` working.

stdevAlDen, 2019-03-27:
> > @stdevAlDen if this is to validate device TRNG is it possible to use some existing code for this ?
> 
> @olemis sure, but I recommend to make (and maintain) a fork of https://github.com/trezor/rng-test and add [this feature](https://github.com/trezor/python-trezor/blob/master/tools/rng_entropy_collector.py) to hardware-wallet-go
> 
> PD: I'm having problems to make `python-trezor` and/or `rng-test` working.

@gz-c olemis Olemis told me that your opinion here is very important, at the same moment that someone makes a final decision, I can start working on that alternative. 
Other alternative(as I can see) is to continue analyzing what I am doing wrong when using the Trezor tools as they are and maybe solve any problem if any.

stdevAlDen, 2019-03-28:
[The firmware build tooling in trezor rng-test project do not link against the compiler predefined startup file](https://github.com/trezor/rng-test/blob/229f60a3eb7b0082f7fda28ccc598957903d643a/dump-stm32/Makefile.include#L57), but I can no find a custom file [as in our project](https://github.com/skycoin/hardware-wallet/blob/c305a2d6de1335070c60cedf259f45c27a963ffd/tiny-firmware/startup.s), this look a bit strange.

PD:
[-nostartfiles
Do not use the standard system startup files when linking. The standard system libraries are used normally, unless -nostdlib, -nolibc, or -nodefaultlibs is used.](https://gcc.gnu.org/onlinedocs/gcc/Link-Options.html)

stdevAlDen, 2019-03-28:
> @gz-c olemis Olemis told me that your opinion here is very important, at the same moment that someone makes a final decision, I can start working on that alternative.
> Other alternative(as I can see) [...]

[Another alternative may be to use our defined messages to get the entropy from the device to the host machine.](https://github.com/skycoin/hardware-wallet-protob/blob/master/protob/messages/messages.proto#L308-L324)

stdevAlDen, 2019-03-28:
[python-trezon](https://github.com/trezor/python-trezor) can not be used as is, this give this error:
```bash
Traceback (most recent call last):
  File "./rng_entropy_collector.py", line 36, in <module>
    main()
  File "./rng_entropy_collector.py", line 19, in main
    raise e
  File "./rng_entropy_collector.py", line 16, in main
    client = TrezorClient(get_transport(), ui=ui.ClickUI())
  File "/home/user/.local/lib/python3.7/site-packages/trezorlib/transport/__init__.py", line 135, in get_transport
    return next(iter(enumerate_devices()))
StopIteration
```

maybe `enumerate_devices` is based on vendor id or some other specific trezor info.

olemis, 2019-03-29:
@stdevAlDen this is not an error

olemis, 2019-04-03:
what needs to be done with the data generated by `check-trng` ?

olemis, 2019-04-11:
fixed in 9742161

PR #157 (2019-04-11): Validate trng fixes #105
Fixes #105

To be able to run this tests this pr is required: https://github.com/skycoin/hardware-wallet-go/pull/87

Changes:
- Add rules to validate the `TRNG`.
- Make the button confirmation for get entropy optional at build time `DISABLE_BUTTON_CONFIRMATION_TO_GET_ENTROPY`

Does this change need to mentioned in CHANGELOG.md?
yes

Requires changes in protobuff specs?
yes
https://github.com/skycoin/hardware-wallet-protob/issues/32

related to PR skycoin/hardware-wallet-protob# for issue skycoin/hardware-wallet-protob#

Requires testing
no

PD: you can check the attach [... cut]
closed 2019-04-11 by olemis