skycoin/hardware-wallet #110: Error in function documentation.

stdevAlDen, 2019-03-01:
**Describe the bug**
There is some inconsistency between function documentation and function behavior.

**Environment information:**

- Any

**Steps to Reproduce**

1. Go to [`recover_pubkey_from_signed_message`](https://github.com/skycoin/hardware-wallet/blob/4aa0d5334eadcc6549dcb2fc58f951f549c5928d/skycoin-api/skycoin_check_signature.c#L89) and look at the documentation.
2. As the return value come from [here](https://github.com/skycoin/hardware-wallet/blob/4aa0d5334eadcc6549dcb2fc58f951f549c5928d/skycoin-api/skycoin_check_signature.c#L102).
3. The [return value](https://github.com/skycoin/hardware-wallet/blob/4aa0d5334eadcc6549dcb2fc58f951f549c5928d/skycoin-api/skycoin_check_signature.c#L31) can be 1.

**Actual behavior**
Function documentation do not fit all the possible values, even it's not clear if a 5 return is possible.

**Expected behavior**
Documentation should match with the implementation.

YuraYelisieiev, 2019-03-11:
The return value that came from here[https://github.com/skycoin/hardware-wallet/blob/4aa0d5334eadcc6549dcb2fc58f951f549c5928d/skycoin-api/skycoin_check_signature.c#L102](url), is 1 if there was an error during verification. So a 5 return is not possible. Just need to change 5 to 1 in the documentation.

jdknives, 2019-07-12:
@stdevAlDen @YuraYelisieiev this file has been moved or deleted on the recent branch and I could not find it with a grep. Is this still a valid issue?

stdevAlDen, 2019-07-15:
> @stdevAlDen @YuraYelisieiev this file has been moved or deleted on the recent branch and I could not find it with a grep. Is this still a valid issue?

@jdknives looks as deprecated but in any case [this return value](https://github.com/skycoin/hardware-wallet/blob/862f0bd5db4988bc6d6bdef0af652f66917a9794/tiny-firmware/firmware/fsm_impl.c#L260) need to be checked.

PR #312 (2019-07-17): fix usage of skycoin_address_from_pubkey ref #110
Fixes #110

Changes:	
- Fix usage of `fsm_getKeyPairAtIndex`
- Fix usage of `skycoin_address_from_pubkey`

Does this change need to mentioned in CHANGELOG.md?	
no	

Requires testing	
no
closed 2019-07-17 by olemis