skycoin/hardware-wallet #111: Decimals are not being displayed when previewing a transaction

Senyoret1, 2019-03-03:
**Describe the bug**
When creating a transaction, the device screen only shows the whole part of the amount of coins the user wants to send. For example, if when trying to send 1.5 coins, the screen simply shows 1.

**Environment information:**

- OS: all
- Platform: all
- Firmware Version: develop branch, updated to  4aa0d53.

**Steps to Reproduce**
Create a transaction sending fractions of a coin.

**Actual behavior**
As described at the beginning.

**Expected behavior**
The hardware wallet should show decimals.

**Additional context**
Decimals worked correctly before, so there was probably a problem with some commit added to the develop branch after https://github.com/skycoin/hardware-wallet/pull/93 was merged.

stdevStark, 2019-03-21:
How can I use SkyWallet from desktop wallet ?

stdevIvr, 2019-03-22:
[image]

It works for me.....

stdevIvr, 2019-03-22:
Ohh, I see. My case works fine, but when I tried 1.5 I only got 1...

olemis, 2019-03-23:
fixed by bd9320e

closed 2019-03-23 by olemis

Senyoret1, 2019-03-24:
> 
> 
> How can I use SkyWallet from desktop wallet ?

@stdevStark sorry for the late response.

For compatibility with the hardware wallet to be enabled, you must rebuild the front-end running `npm run build-prod-hw` in `src/gui/static` and then rebuild the Electron executable.

PR #150 (2019-03-23): fixes #111 Print 1.5 coins correctly
Fixes #111

Changes:
- Correct text when coins are 1.5
- Verify work for different fractions of coins

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes

Examples

4.9023 SKY = 4902300 droplets
9.703  SKY = 9703000 droplets
3.23  SKY = 3230000 droplets
2.01  SKY = 2010000 droplets
1.5  SKY = 1500000 droplets
0.1  SKY = 100000 droplets
commit bd9320ee (2019-03-23): Merge pull request #150 from simelo/stdevStark_t111_sprintcoins_decimals

fixes #111 Print 1.5 coins correctly