skycoin/hardware-wallet #115: mnemonic passphrase feature requires missing method

gz-c, 2019-03-06:
https://github.com/trezor/trezor-crypto/blob/b7e99aa76c36417b797add2addba03121b758e66/bip39.c#L188

The method linked above is required to use the passphrase feature of mnemonic generation.

This method was removed from our source tree. Restore the method.  The bip39 library should probably be restored in full, from the trezor sources. Unless there is a good reason to modify it.

olemis, 2019-03-06:
This is already in source tree 

```sh
$ grep -nr mnemonic_to_seed .
./tiny-firmware/vendor/skycoin-crypto/trezor-tools/bip39.h:40:void mnemonic_to_seed(const char *mnemonic, const char *passphrase, uint8_t seed[512 / 8], void (*progress_callback)(uint32_t current, uint32_t total));
./skycoin-api/trezor-tools/bip39.h:40:void mnemonic_to_seed(const char *mnemonic, const char *passphrase, uint8_t seed[512 / 8], void (*progress_callback)(uint32_t current, uint32_t total));
```

olemis, 2019-03-06:
`./tiny-firmware/vendor/skycoin-crypto/trezor-tools` is a symlink to `./skycoin-api/trezor-tools` .

olemis, 2019-03-07:
`bip39.c` file is missing

gz-c, 2019-03-07:
This is not assigned to the release milestone so does not need action at this time

olemis, 2019-03-08:
fixed by d1b8fa3

PR #118 (2019-03-08): fixes #115 Restore bip39 library implementation
Fixes #115 

Changes:
- Restore trezor tools `bip39.h` .

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes
commit d1b8fa3c (2019-03-08): Merge pull request #118 from simelo/olemis_t115_restore_bip39

fixes #115 Restore bip39 library implementation
closed 2019-03-08 by olemis