skycoin/hardware-wallet #119: Generating mnemonic with 17 words

stdevIvr, 2019-03-09:
**Describe the bug**
The firmware for the wallet is allowing to work with mnemonics of 17 words, when it should accept only 12 or 24 words.

**Environment information:**

- OS: Linux
- Platform: Linux 4.15.0-46-generic x86_64
- Firmware Version: 1cbb340

**Steps to Reproduce**
* Go to the [skycoin/hardware-wallet](https://github.com/skycoin/hardware-wallet/) repository and run the emulator
* Go to the latest commit in the head (**git checkout 7a4164**)
* Go to [my repo](https://github.com/simelo/skycoin-hardware-wallet-js/tree/stdevIvr_t38_tests_with_mocha_and_chai)
* Install all the dependencies (**npm install**)
* Go go the latest commit (**git checkout 26b075**)
* Run the message's test: **./node_modules/.bin/mocha ./test/messages-test.js**
* Press the OK button in the emulator (right arrow) until the test ends.
* See that the tests are failing because the firmware is accepting 17 words mnemonics.

**Actual behavior**
The tests are failing because the firmware is accepting 17 words mnemonics

**Expected behavior**
The tests should pass properly. In other words, the firmware must only accept 12 or 24 words mnemonics.

jdknives, 2019-03-09:
The firmware should accept 12, 15, 18, 21, 24. The 12/24 limitation will be handled by the UI only. We should not modify the BIP39 from Trezor. Any modification is a violation of the BIP39 spec.

olemis, 2019-03-09:
> The firmware should accept 12, 15, 18, 21, 24. The 12/24 limitation will be handled by the UI only. We should not modify the BIP39 from Trezor. Any modification is a violation of the BIP39 spec.

for the record , inherited code only supports mnemonic lengths of 12 , 18, and 24 .

jdknives, 2019-03-09:
The orginal Trezor code only supports 12, 18 and 24?

olemis, 2019-03-09:
Neither the [crypto API](https://github.com/trezor/trezor-crypto/blob/master/bip39.c#L119-L121) nor [firmware messages](https://github.com/trezor/trezor-mcu/blob/44bb5864d42e95bb5c15a351f21792df83f4fc61/firmware/fsm_msg_common.h#L349).

Nonetheless this issue is not about implementing specific word count values , but about a critical bug in firmware that leads to generating 24 words mnemonic if a value different than 12 is specified.

olemis, 2019-03-09:
Thanks @stdevIvr for reporting this . The fact that you found this is a proof of the good work improving the test suite . Has it been completed ? Should you find further bugs please report them to decide when to fix them . Low severity bugs should not make it into milestone 1 .

olemis, 2019-03-09:
[...]
> a critical bug in firmware that leads to generating 24 words mnemonic if a value different than 12 is specified.

Expected result is sending failure message .

gz-c, 2019-03-09:
ok 12,18,24 is fine if libbip39 does it

gz-c, 2019-03-09:
https://github.com/trezor/trezor-crypto/blob/master/bip39.c

see mnemonic_generate

12,15,18,21,24 words are allowed, corresponding to 128,170,192,224,256 entropy bits

olemis, 2019-03-10:
fixed in af9523f

PR #122 (2019-03-10): [firmware] fixes #119 - Validate mnemonic count in firmware 

Fixes #119 

Changes:
- Check that word count in `GenerateMnemonic` message is either 12 or 24 . 

Known issues

Other valid values may be added later .

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes

Test case added in `test_fsm.c`.
commit af9523f3 (2019-03-10): Merge pull request #122 from simelo/olemis_t119_mnemonic_words_12_24

[firmware] fixes #119 - Validate mnemonic count in firmware
closed 2019-03-10 by olemis