skycoin/hardware-wallet #121: Use salted entropy pool mixer

gz-c, 2019-03-10:
Entropy sourced from external and internal should be mixed into an entropy pool mixer which should be salted with some value(s).

See https://github.com/skycoin/skycoin/blob/develop/src/cipher/secp256k1-go/secp256_rand.go#L46 for a reference of how to do the mixing.

Unsure where to get salted values in the embedded device.  Possible sources:

- Clock/systime (if available on the stm32 chip)
- Some global incrementing program counter
- Any variable source of data available to the device other than the rng itself

Some links on the reasoning for this:
https://crypto.stackexchange.com/questions/17658/mixing-entropy-sources-by-xor

gz-c, 2019-03-10:
Looking at the datasheet (https://www.mouser.sg/datasheet/2/389/stm32f205rb-956262.pdf) I see these possible salt sources:

* 96-bit unique ID
* "On reset the 16 MHz internal RC oscillator is selected as the default CPU clock. The
16 MHz internal RC oscillator is factory-trimmed to offer 1% accuracy." - The clock speed if we can query it
* real-time clock (RTC)
* temperature sensor
* SysTick timer
* General-purpose timers (TIMx)

stdevAlDen, 2019-03-13:
> Entropy sourced from external and internal should be mixed into an entropy pool mixer which should be salted with some value(s).
>
> See [...]

@gz-c | @olemis Can you pls give my a hint about a [`SumSHA256`](https://github.com/skycoin/skycoin/blob/develop/src/cipher/secp256k1-go/secp256_rand.go#L23) alternative from [here](https://github.com/skycoin/hardware-wallet/blob/master/skycoin-api/tools/sha2.h#L100-L106) (or maybe some other place)?

`SumSHA256` is used [here](https://github.com/skycoin/skycoin/blob/develop/src/cipher/secp256k1-go/secp256_rand.go#L52)

olemis, 2019-03-13:
https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/reset.c#L85-L89

stdevAlDen, 2019-03-13:
> @stdevAlDen see
> 
> https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/reset.c#L85-L89

Is this the code I need to delete/replace or did I misunderstand the issue purpose?

olemis, 2019-03-14:
>     * 96-bit unique ID

@gz-c we are having issues reading device `UID` in the firmware . Indeed we tried to do so for creating a default label and ended up doing what Trezor does , generate random ID .

olemis, 2019-03-19:
> * 96-bit unique ID
> 

This one does not look like a good source of entropy since it has null variance , and is meant to be rendered in wallet GUI . Moreover reading from firmware leads to memory protection fault .

> * "On reset the 16 MHz internal RC oscillator is selected as the default CPU clock. The
>   16 MHz internal RC oscillator is factory-trimmed to offer 1% accuracy." - The clock speed if we can query it
> 

it seems variations cannot be read with any accuracy at all . 

> * temperature sensor
> 

@stdevAlDen working on this 

> * real-time clock (RTC)
> 
> * SysTick timer
> 
> * General-purpose timers (TIMx)

these can be used

olemis, 2019-03-19:
> > * "On reset the 16 MHz internal RC oscillator is selected as the default CPU clock. The
> >   16 MHz internal RC oscillator is factory-trimmed to offer 1% accuracy." - The clock speed if we can query it
> 
> it seems variations cannot be read with any accuracy at all .

we'll use it though

olemis, 2019-04-07:
fixed by c760a28

PR #136 (2019-04-07): Use salted entropy pool mixer fixes #121
Fixes #121 

Changes:
- Remove Shanon entropy level verification.
- Merge entropy in a pool mixer.

Does this change need to mention in CHANGELOG. MD?
yes

Requires testing
yes

PD: The use of more entropy sources was postponed
https://github.com/skycoin/hardware-wallet/issues/141
closed 2019-04-07 by olemis