skycoin/hardware-wallet #123: Deactive the method for signing generic messages

Senyoret1, 2019-03-11:
The method for signing generic messages (the one called with `devSkycoinSignMessage` in the JS library) should be deactived for V1, since it is not used anywhere.

**Describe alternatives you've considered**
https://github.com/skycoin/hardware-wallet/issues/77 , but deactivating it is better for V1.

olemis, 2019-04-04:
No need to disable since code in `develop` branch protects message signing by pin code .

olemis, 2019-04-04:
@Senyoret1 if you try it and get a different result let me know . Tested manually with hw-go library .

closed 2019-04-04 by olemis

Senyoret1, 2019-04-04:
@olemis It is true that the pin code is requested before being able to sign a generic message, but there are problems with that. The user may decide not to use a pin code, and in that case the device must continue being safe. Also, if the user recovers an old seed and then creates a pin code, the device stays without pin for some time, time in which it would be vulnerable.

The Trezor devices always ask for confirmation defore signing generic messages. most likelly for the previous reasons.

olemis, 2019-04-05:
> The Trezor devices always ask for confirmation defore signing generic messages. most likelly for the previous reasons.

This is what will be done . See #77 .

olemis, 2019-04-05:
I agree with all your arguments above .

Senyoret1, 2019-04-07:
Well, asking for confirmation is better, this issues was created just in case deactivating the method was faster than asking for confirmation. Now that this issue is closed, #77 should be added to V1.

Senyoret1, 2019-04-07:
Hmm, I dont know why Github unassigned @stdevStark with my last comment :frowning:

olemis, 2019-04-07:
> Hmm, I dont know why Github unassigned @stdevStark with my last comment 😦

nothing you did yourself , everything will be fine once @stdevStark will deal with account configuration issues .

stdevStark, 2019-04-07:
@olemis I did 2FA . I need to do something to join ?