skycoin/hardware-wallet #124: Hardware wallet default language

therealssj, 2019-03-11:
**Describe the bug**
If empty string is passed as a call to `ApplySettings` use `English` as default instead of showing empty

[image]

**Environment information:**
- Any

**Steps to Reproduce**
use js or go lib and call `ApplySettings` with language as `""`

**Actual behavior**
Shows empty string

**Expected behavior**
Show default language

olemis, 2019-03-14:
should be fixed as a side effect for completing skycoin/hardware-wallet-js#59 .

olemis, 2019-03-14:
fixed in 4a00e78

closed 2019-03-14 by olemis

Senyoret1, 2019-03-17:
@olemis , apparently the problem is still present.

stdevIvr, 2019-03-18:
@Senyoret1 I just tested it and it works. Are you on `develop` branch? Is it the latest commit??

olemis, 2019-03-18:
> @Senyoret1 I just tested it and it works. Are you on `develop` branch? Is it the latest commit??

same with the emulator ... will test in the device later today .

Senyoret1, 2019-03-18:
@stdevIvr I think you are right, the change was made to solve what was originally posted. I was thinking more about what I was talking about in https://github.com/skycoin/hardware-wallet-js/pull/62#pullrequestreview-214528018 . I created a new issue with the details: https://github.com/skycoin/hardware-wallet/issues/135

PR #128 (2019-03-14): [firmware] fixes #124 - Verify language and set to English if empty
Fixes #124 

Changes:
- Added a function that verifies the language provided by the user in `applySettings` and set it to `English` if empty

Does this change need to mentioned in CHANGELOG.md?
yes

Requires testing
yes

Comments about testing , should you have some
commit 4a00e788 (2019-03-14): Merge pull request #128 from simelo/stdevIvr_t124_wallet_default_language

[firmware] fixes #124 - Verify language and set to English if empty
commit 547a4170 (2019-03-14): [wallet] fixes #124 - Verify language and set to English if empty