skycoin/hardware-wallet #131: [emulator] Unexpected code on the screen after backup

stdevIvr, 2019-03-17:
**Describe the bug**
There are some characters on the emulator's screen after calling the `devBackupDevice` function.

**Environment information:**

- OS: Linux
- Platform: Linux 4.15.0-46-generic #49~16.04.1-Ubuntu SMP Tue Feb 12 17:45:24 UTC 2019 x86_64
- Firmware Version: Don't know...

**Steps to Reproduce**

1. Clone [this](https://github.com/skycoin/hardware-wallet) repo (develop branch)
2. Run `make clean` and then `make run-emulator`
3. Clone [this](https://github.com/simelo/skycoin-hardware-wallet-js) repo (at master)
4. Set to `testTransactionSign = false` and `testBackup = true`
5. Run `node test_codes.js`
6. Either you get error or success, you will see after the whole process, the error on the screen.

**Actual behavior**
Look at this picture:

[image]

**Expected behavior**
There is no reason for that sequence to appear on the screen.

**Additional context**
The sequence that appears on the screen is different every time, at least for me.

stdevAlDen, 2019-04-11:
@stdevIvr this is the device id, will be replaced by a the device label when you set up one.
@jdknives please consider closing this

olemis, 2019-04-29:
UUID read from device , fixed in b469846

PR #203 (2019-04-29): Read device UUID ref #202
Fixes #202
closes #100
closes #131 (@stdevIvr these number are the device uuid, coming physical device id with this pr, a random one in emulator, so can be different in emulator)

Changes:
- Read uuid from physical device.
- Please do not miss [this change](https://github.com/skycoin/hardware-wallet/pull/203/commits/a094382a48bc4f0dba3d5e690b3d042030a9fe97). ref #100 

Does this change need to mentioned in CHANGELOG.md?
yes

Requires changes in protobuf specs?
no

Requires testing
no
closed 2019-04-29 by olemis