skycoin/hardware-wallet #133: Allow disable the PIN code

Senyoret1, 2019-03-17:
**Feature description**
Once the user activates the PIN code, the only way to disable it seems to be calling the function that allows to change the PIN (`devChangePin` in the JS library) and send null as the new PIN. However, this method is not good for the UX (the device shows the matrix for entering the new PIN).

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
There should be a function that for disabling the PIN without using "tricks"

olemis, 2019-04-04:
what if a `Clear Pin` button is shown in desktop wallet GUI ? On-the-wire the (only?) way to specify no pin is by setting it to null .

Senyoret1, 2019-04-04:
Oh, I did not know it was possible to delete the pin code that way. I just tested and it works. With that I can implement the option, but it will have room for improvement.

If I just send `null`, the device will request confirmation for changing the pin, but in the Trezord devices request confirmation for deleting it, so I think they use a specific message for removing the pin, or at least a param in the message for changing the pin.

For the user the difference would just be the confirmation message in the device, so it is not something prioritary, but it would be good for the future.

jdknives, 2019-07-17:
@Senyoret1 this has been implemented already as far as I can see. Is that correct?

Senyoret1, 2019-07-17:
Yes, I'm closing it

closed 2019-07-17 by Senyoret1