skycoin/hardware-wallet #134: Screen text cut after firmware upload

therealssj, 2019-03-18:
**Describe the bug**
Text not fully displayed after firmware update.
[image]

**Environment information:**

- OS: any

**Steps to Reproduce**
Steps to reproduce the behavior:

1. Update firmware

**Actual behavior**
text cut around borders

**Expected behavior**
un-cut text

**Possible implementation**
Shorten message to just say `wallet` instead of `Skycoin wallet`

stdevAlDen, 2019-03-18:
@therealssj maybe it's a duplicate of https://github.com/skycoin/hardware-wallet/issues/83

therealssj, 2019-03-18:
oh yeah. 
@stdevAlDen maybe update that issue with the image? Closing this one

closed 2019-03-18 by therealssj

stdevAlDen, 2019-03-18:
> maybe update that issue with the image? Closing this one

sorry I no understand you, #83 currently have an img, so what do you mean?

stdevAlDen, 2019-03-18:
> > maybe update that issue with the image? Closing this one
> 
> sorry I no understand you, #83 currently have an img, so what do you mean?

ok, closed...

PR #181 (2019-04-14): Fixes #176 Update firmware message
Fixes #176 
Fixes #83 
Fixes #134 

Changes:
- Update firmware message in order to show a message according to the size of the screen.

Does this change need to mentioned in CHANGELOG.md?
yes
commit b894b638 (2019-04-14): Merge pull request #181 from simelo/stdevMac_t176_update_firmware_message

Fixes #176 Fixes #83 Fixes #134 - Skycoin wallet => Skywallet