skycoin/hardware-wallet #135: Do not change the language if it is not necessary

Senyoret1, 2019-03-18:
**Describe the bug**
When the [devApplySettings](https://github.com/skycoin/hardware-wallet-js/blob/master/DOCUMENTATION.md#devapplysettings) function of the JS library is called, it is possible to send `null` in the `usePassphrase` and `deviceLabel` parameters to avoid making changes to that configuration elements. For example, I can send `null` in the `usePassphrase` param and `"test"` in the `deviceLabel` param and the hardware wallet will only ask if I want to change the device label. However, it is not possible to do this with the language.

The `devApplySettings` function does not accept a parameter for the language, so it always sends an empty string and the device always asks if the lenguaje should be changed to English. That is ok, but if the library code is modified to send always `null` it shoud do nothing (the lenguaje should not be changed and the device should not ask for confirmation), just as it happens with the passphrase protection and the label. However, what happens is that the device responds as if the operation was performed correctly but in reality no change is made and confirmation is not requested.

This change is necessary because, if it is not done, the user will always be asked to confirm a language change that was not requested, which can be perceived as an operating error.

**Environment information:**

- OS: all
- Platform: all
- Firmware Version:develop branch cb522ac (last commit at this time)

olemis, 2019-03-22:
fixed in 301feff
@Senyoret1 please try again . Feel free to reopen this issue if you still notice this issue.

olemis, 2019-03-22:
> The `devApplySettings` function does not accept a parameter for the language, so it always sends an empty string and the device always asks if the lenguaje should be changed to English

JFTR , support for language in js lib is beyond the scope of this repository .

closed 2019-03-22 by olemis

Senyoret1, 2019-03-23:
@olemis It is working well now :+1: . During the tests I found another problem, but it is different, so put it in https://github.com/skycoin/hardware-wallet/issues/146

PR #138 (2019-03-22): [wallet] refs #135 - Disable confirmation if no language provided
Fixes #135 

Changes:
- Disabled the confirmation for set the language when an empty string is provided.

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes
commit 301feff0 (2019-03-22): Merge pull request #138 from simelo/stdevIvr_t135_do_not_change_language_if_not_necessary

[wallet] fixes #135 - Disable confirmation if no language provided