skycoin/hardware-wallet #141: Add more entropy sources

stdevAlDen, 2019-03-21:
**Describe the bug**
The entropy pool mixer need more entropy sources:
- Clock/systime (if available on the stm32 chip)
- Some global incrementing program counter
- Any variable source of data available to the device other than the rng itself

**Environment information:**

Any

**Steps to Reproduce**
Go to the [parent issue](https://github.com/skycoin/hardware-wallet/issues/121#issuecomment-471236023) for more details **it's very important**.

**Actual behavior**
Having few entropy sources.

**Expected behavior**
Add more entropy sources.

**Additional context**
https://github.com/skycoin/hardware-wallet/issues/121

**Possible implementation**
Can be tricky, not support for adc stm32 series f2 from libopencm3 https://github.com/libopencm3/libopencm3/commit/3eaeaf693c02eba7aee0119d37c1dff3b16b4765

olemis, 2019-03-23:
I will edit the description of this issue since it is not about entropy sources , but salt .

olemis, 2019-03-23:
> Can be tricky, not support for adc stm32 series f2 from libopencm3 [libopencm3/libopencm3@3eaeaf6](https://github.com/libopencm3/libopencm3/commit/3eaeaf693c02eba7aee0119d37c1dff3b16b4765)

We'll have to be done from scratch then .

gz-c, 2019-03-26:
I see the important comment about "variable" is crossed out:

For "variable", there are a few meanings:
 1. variable between devices (such as device UUID)
 2. variable between device runs (at init time)
 3. variable over time (after init, value continues to change)

On PCs:
1. example is env vars
2. example is process id and unix time
3. example is unix time

If a value is only variable in (3) and not (1) or (2) (i.e. every device has the same value on init) then it is not useful unless we periodicially mix it in to the entropy pool.

If there is no/low cost to sampling a variable (3) value, you can sample it every time new entropy is requested. Then (1) and (2) only need to be sampled on init or the first request

gz-c, 2019-03-26:
And, for difficult to obtain salts, I had said this:

Use as many variable sources of data as you can, **within reason (i.e. not exceptionally difficult to acquire)**

gz-c, 2019-03-26:
An example of something "within reason" is a simple function call or memory read

olemis, 2019-04-07:
@gz-c other possible salt sources

| Address range  |  Memory-mapped registers |
|----------------|--------------|
| 0xE000E010-0xE000E01F | System timer |
| 0xE000E100-0xE000E4EF | Nested vectored interrupt controller |
| 0xE000ED00-0xE000ED3F | System control block |
| 0xE000ED90-0xE000ED93 | Memory protection unit |
| 0xE000EF00-0xE000EF03 | Nested vectored interrupt controller |

olemis, 2019-04-07:
most of these would require privileged mode access

stdevStark, 2019-04-16:
@olemis this needs that I have a device .

olemis, 2019-04-16:
right

olemis, 2019-05-02:
fixed in cde6940

PR #197 (2019-05-02): fixes #141 Salt entropy pool with values of MCU registers
Fixes #141

Changes:
- Salt entropy pool with values of LR, SP, and PC registers

Does this change need to mentioned in CHANGELOG.md?
no

Requires changes in protobuff specs?
no

Requires testing
no
requires running on the physical device
commit cde69406 (2019-05-02): Merge pull request #197 from simelo/olemis_t141_salt_mcu_reg

fixes #141 Salt entropy pool with values of MCU registers
closed 2019-05-02 by olemis