skycoin/hardware-wallet #146: The device label is sometimes deleted when calling ApplySettings

Senyoret1, 2019-03-23:
**Describe the bug**
If the device has a label and you call `ApplySettings` setting the passphrase protection to any value, but do not set a label, the device label is replaced by the default label. This does not happens with the passphrase protection: if `ApplySettings` is called for setting a new label but without specifying a value for the passphrase protection, the current passphrase configuration is not changed.

**Environment information:**

- OS: All.
- Platform: All.
- Firmware Version: Develop branch updated to e3f505e (the lastd commit at this time).

**Steps to Reproduce**
Steps to reproduce the behavior:

1. Call `ApplySettings` to set a new label, without indicating any value for the passphrase protection.
2. Call `ApplySettings` to set the passphrase protection to false, without indicating any value for the label.
3. Check the device, the label set on the first step will be gone.

**Expected behavior**
The label should not be erased, as the function call only asked for changes in the passphrase protection.

**Additional context**
As the passphrase protection will not be enabled in V1, solving this issue is not needed for V1.

olemis, 2019-03-23:
@Senyoret1 we need more info before dealing with this . How did you send those messages ? Using hw-js library ? When you say `do not set label` what value exactly are you specifying ?

olemis, 2019-03-23:
> * Call `ApplySettings` to set a new label, without indicating any value for the passphrase protection.
> 
> * Call `ApplySettings` to set the passphrase protection to false, without indicating any value for the label.

Protobuffer forces optional values to be `null` . At least that's what it says in `skycoin.js` after generating protobuffer code for js.

```js
$root.ApplySettings = (function() {

    /**
     * Properties of an ApplySettings.
     * @exports IApplySettings
     * @interface IApplySettings
     * @property {string|null} [language] ApplySettings language
     * @property {string|null} [label] ApplySettings label
     * @property {boolean|null} [usePassphrase] ApplySettings usePassphrase
     * @property {Uint8Array|null} [homescreen] ApplySettings homescreen
     */

    /**
     * Constructs a new ApplySettings.
     * @exports ApplySettings
     * @classdesc Request: change language and/or label of the device
     * @start
     * @next Success
     * @next Failure
     * @implements IApplySettings
     * @constructor
     * @param {IApplySettings=} [properties] Properties to set
     */
    function ApplySettings(properties) {
    ...
    }
```

If not set values would be undefined . If set to empty string it seems that's not the right way to do it . Could you please try again , setting these to `null` and share your results here ?

Senyoret1, 2019-03-24:
I tested using the js library. For the first step I used `deviceWallet.devApplySettings(null, 'label')` and for the second one `deviceWallet.devApplySettings(false, null)`, so with not sending a value I menan using `null`.

olemis, 2019-03-27:
fixed by c305a2d

PR #152 (2019-03-27): Delete label by accident ref #146
Fixes #146

Changes:
- Do not delete(restore to a default value) label if it is not have a value in applySettings.

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
yes
commit c305a2d6 (2019-03-27): Merge pull request #152 from simelo/stdevAlDen_t146_delete_label_by_accident

Delete label by accident fixes #146
closed 2019-03-27 by olemis