skycoin/hardware-wallet #151: Improve validations of input data

Senyoret1, 2019-03-24:
**Describe the bug**
This issue is a response to https://github.com/skycoin/hardware-wallet-js/pull/75#issuecomment-475463787

Currently it is a little difficult to predict what the hardware wallet reaction will be when unexpected data is sent to it. In some cases, when sending invalid data to it, the hardware wallet does nothing but responds as if the operation had been performed correctly or perform operations with results that are difficult to predict. Some cases I have detectd (mosly using the JS library) are:

- If the `ApplySettings` message is called with `usePassphrase` set to `null`, the response is that the operation finished correctly, but in fact nothing happens.

- When assigning a PIN to the device for the first time or changing the current one, if `null` is sent as the new PIN the response is that the operation finished correctly, but in fact nothing happens. If a string sent instead of `null`, the response is that the operation finished correctly and, in fact, the device start asking for the PIN. This is specially strange because the hardware wallet accepts as valid PIN any string as long as it has the same number of characters as the string used when assigning the PIN.

- It is possible to send any value (even an object) in the parameter of `devAddressGen` funsction that indicates the initial index, and it will return addresses starting from the one with index 0.

- When calling `devSkycoinTransactionSign` it is possible to avoid indicating the index of the address that must be used to sign an input. Including that param shold be mandatory. I do not know which address the device uses to make the signature in that case, although it could be the first one.

**Environment information:**

- OS: Tested using Windows, but should be the same in other operative systems.
- Platform: Windows 10 x86_64.
- Firmware Version: Develop branch updated to 39229d6.

**Steps to Reproduce**
Call the functions mentioned above with the example parameters.

**Actual behavior**
As mentioned above.

**Expected behavior**
As far as possible, the hw wallet should not be performing actions on arbitrary data.

**Additional context**
Most of the tests were done with the JS library.

**Possible implementation**

stdevHan, 2019-03-30:
@Senyoret1 The validations defined for the `devSkycoinTransactionSign`, must be defined in the libjs, for the firmware those validations are already defined. For the rest they are already repaired in the PR https://github.com/skycoin/hardware-wallet/pull/153.

olemis, 2019-04-04:
@Senyoret1 I am doing some tests and add a few more items that need testing .

olemis, 2019-04-08:
>     * If the `ApplySettings` message is called with `usePassphrase` set to `null`, the response is that the operation finished correctly, but in fact nothing happens.

@Senyoret1 about this I added a test for firmware in 4ad1be8 and it succeeds .

olemis, 2019-04-09:
@Senyoret1 apply settings is working as expected , according to tests in firmware . The storage is updated as expected , etc . Nonetheless I noticed this 

```sh

```

> * When assigning a PIN to the device for the first time or changing the current one, if `null` is sent as the new PIN the response is that the operation finished correctly, but in fact nothing [...]

`devChangePing` command is not correct . This has to be implemented in hw-js .

Senyoret1, 2019-04-09:
> `devChangePing` command is not correct . This has to be implemented in hw-js .

The first part of this is not true anymore (I do not know if because of a change in the firmware or a mistake of mine while checking), when sending `null` the pin is erased. However, the part about the random character string is still true. In fact, I just checked it using the daemon and the problem hapened, so I think it should not be solved in the js library, but in the firmware.

stdevStark, 2019-04-09:
> The first part of this is not true anymore (I do not know if because of a change in the firmware or a mistake of mine while checking), when sending `null` the pin is erased.

`devChangePin` is not right . The function to assemble the message it uses has `mnemonic` as parameter . The protobuf message only has `remove` bool variable.

olemis, 2019-04-10:
>     * When calling `devSkycoinTransactionSign` it is possible to avoid indicating the index of the address that must be used to sign an input. Including that param shold be mandatory. I do not know which address the device uses to make the signature in that case, although it could be the first one.

This is not a bug . This is for signing inputs of transactions spending coins owned by an address not managed by the physical Skywallet device .

olemis, 2019-04-10:
> `devChangePin` is not right . The function to assemble the message it uses has `mnemonic` as parameter . The protobuf message only has `remove` bool variable.

This was in js ... but also buggy in firmware . Tested now (in firmware) and should be ok.

Senyoret1, 2019-04-11:
> This is not a bug . This is for signing inputs of transactions spending coins owned by an address not managed by the physical Skywallet device .

If that is the case, we should document somewhere how to use that feature. But it is not prioritary.

olemis, 2019-04-12:
>     * It is possible to send any value (even an object) in the parameter of `devAddressGen` funsction that indicates the initial index, and it will return addresses starting from the one with index 0.

This is working as expected in firmware . See #175 .

@stdevIvr @stdevStark further changes should be implemented in skycoin/hardware-wallet-js#97 .

olemis, 2019-04-12:
>     * When calling `devSkycoinTransactionSign` it is possible to avoid indicating the index of the address that must be used to sign an input. Including that param shold be mandatory. I do not know which address the device uses to make the signature in that case, although it could be the first one.

@stdevMac is writing tests for this .

olemis, 2019-04-15:
@stdevMac the only thing that's left is transaction tests .

stdevMac, 2019-04-15:
Working on tests

denisacostaq, 2019-04-24:
@stdevMac there is a problem here to make the test working?
https://github.com/skycoin/hardware-wallet/blob/416425751f37a9f52de921b0481f7bec29d32827/tiny-firmware/firmware/fsm_impl.c#L366

How are you trying to bypass this request?

stdevMac, 2019-04-24:
@stdevAlDen i don't get any error in this request.

I get an error while i use this function https://github.com/skycoin/hardware-wallet/blob/416425751f37a9f52de921b0481f7bec29d32827/tiny-firmware/firmware/fsm_impl.c#L197

specifically in this line => https://github.com/skycoin/hardware-wallet/blob/416425751f37a9f52de921b0481f7bec29d32827/tiny-firmware/firmware/fsm_impl.c#L221
can you check the parameters i pass?

olemis, 2019-05-03:
fixed in 8350dbc

PR #185 (2019-05-03): Improve validations of input data for transaction signing
Fixes #151

Changes:
- Added test cases for transaction sign

Does this change need to mentioned in CHANGELOG.md?
no

Requires changes in protobuff specs?
no

Requires testing
no
closed 2019-05-03 by olemis