skycoin/hardware-wallet #156: SkyWallet branding

olemis, 2019-03-31:
**Feature description**

Reference SkyWallet .

**Is your feature request related to a problem? Please describe.**
The inherited technology is mentioned all over across the project tree .

**Additional context**
noticed after merging #147 

**Possible implementation**

- [ ] rename source files
- [ ] update file contents
- [ ] ensure output binary files are created with correct names

stdevMac, 2019-04-03:
Should I also change trezor in self-generated comments? Example, if I change the file, since the names of `trezor.h` are changed to `skywallet.h` and all its occurrences, then what do I do with the files in which they are autogenerated or signed with the license, which are modified? I delete the trazor comment or do I leave it?

Example here: https://github.com/simelo/skycoin-hardware-wallet/pull/3/files#diff-a84e671dd7e5ccabf6783b42f67d4c13

stdevMac, 2019-04-03:
Which are the other places that i should change?

olemis, 2019-04-10:
fixed in 50000f6

PR #168 (2019-04-10): fixes #156 - Skywallet branding v2
Fixes #156 

Changes:
- rename source files
- update file contents
- ensure output binary files are created with correct names

Does this change need to mentioned in CHANGELOG.md?
no

Requires testing
no
commit 50000f67 (2019-04-10): Merge pull request #168 from simelo/stdevMac_t156_skywallet_branding

fixes #156 - Skywallet branding 2nd round
closed 2019-04-10 by olemis