skycoin/hardware-wallet #159: Developer contribution guidelines

olemis, 2019-04-02:
Continue developer instructions started in https://github.com/simelo/skycoin-hardware-wallet/blob/stdevIvr_t142_refactor_fsm.c_for_unit_testing/CONTRIBUTING.md .

jdknives, 2019-04-18:
@olemis do you have specific things in mind that you would like added here?

olemis, 2019-04-18:
> @olemis do you have specific things in mind that you would like added here?

There are quite a few items in the [project tree overview](https://github.com/skycoin/hardware-wallet/blob/develop/CONTRIBUTING.md#quick-overview) and only bootloader and firmware covered in the docs.

olemis, 2019-04-24:
@YuraYelisieiev progress ?

YuraYelisieiev, 2019-04-24:
@olemis added guide on how to compile the whole project and now I am working on test writing procedure and policies.

YuraYelisieiev, 2019-04-24:
@olemis done with test cover.

YuraYelisieiev, 2019-05-02:
@olemis Give me pls feedback about the contribution guidelines. What's wrong with the first item in the checklist?

jdknives, 2019-05-02:
I checked the last two ones because they looked good to me. Cannot evaluate the other checkmarks.

olemis, 2019-05-02:
@YuraYelisieiev what's the PR ?

jdknives, 2019-05-02:
https://github.com/skycoin/hardware-wallet/pull/201 @olemis

YuraYelisieiev, 2019-05-02:
@olemis https://github.com/skycoin/hardware-wallet/pull/201

jdknives, 2019-05-06:
Closed by #201

closed 2019-05-06 by jdknives