skycoin/hardware-wallet #166: set pin code not working

therealssj, 2019-04-08:
**Describe the bug**
set pin code returns `Pin Changed` even on entering wrong pin.
On entering correct pin its not set. 

**Environment information:**

- OS: Mac OS X
- Platform: Darwin 18.5.0 x86_64

**Steps to Reproduce**
Steps to reproduce the behavior:

1. use hw-go cli to send `setPinCode`
2. enter `123` for both prompts
3. try again, does not ask for old pin

**Actual behavior**
Returns `Pin Changed` even on entering wrong pin and does not set pin on entering correct pin

**Expected behavior**
Return error on entering wrong pin, set pincode on entering correct pin.

Senyoret1, 2019-04-11:
I checked this and could reproduce the error with the latest firmware.

However, that was not the only error. When calling the function for changing the PIN and using a valid 4-digit PIN, the hardware wallet responded that the 2 pins did not match, but, in fact, the PIN was changed correctly and was required when sending new requests to the device.

This is a important problem, so I think it should have the v1 label.

olemis, 2019-04-11:
The changes merged in ab2e80c should fix this error and provides test cases .

closed 2019-04-11 by olemis

olemis, 2019-04-11:
>     1. use hw-go cli to send `setPinCode`
> 
>     2. enter `123` for both prompts
> 

A few more context about this to clarify a few things , for the record 

Firstly, @therealssj the pin code matrix is meant to be randomized . When entering `123` twice this is what happens under the hood in the firmware :

- hw-go sends `setPinCode`
- hw asksfor confirmation
- user confirms
- hw randomizes digits and displays first matrix e.g.

| 1 | 4 | 6 |
| 7 | 2 | 9 |
| 8 | 5 | 3 |

- hw-go asks for pin code
- user enters `123`
- hw-go send `123` to hw
- hw translates this according to first matrix , e.g. considering the example above hw sees `853`
- hw randomizes digits and displays second matrix e.g.

| 5 | 3 | 8 |
| 4 | 7 | 6 |
| 9 | 1 | 2 |

- hw-go asks for pin code
- user enters `123`
- hw-go send `123` to hw
- hw translates this according to second matrix , e.g. considering the example above hw sees `912`
- compares both pins and they do not match , so it is not set ...

>     3. try again, does not ask for old pin
> 

... and that's the reason why it does not ask for pin for the second time .