skycoin/hardware-wallet #169: The recovery function is not working

Senyoret1, 2019-04-11:
When calling the recovery function (`/recovery` in the daemon and `devRecoveryDevice` in the JS library) and confirming the operation, the device shows very briefly the word that must be written and immediately after that returns to the home screen. Because of this, it is not possible to see the word or continue with the operation.

**Environment information:**

- OS: Windows 10
- Platform: X64
- Firmware Version: 50000f6

**Steps to Reproduce**
Just call `/recovery` (with the daemon) or `devRecoveryDevice` (with the JS library).

**Actual behavior**
After confirming the operation, the device shows very briefly the word that must be written and immediately after that returns to the home screen.

**Expected behavior**
The word that must be written should be shown util the user enters it or cancels the operation.

**Additional context**
I tried doing the same with an old version of the firmware (compiled on March 18) and the problem was not present, the process worked correctly. This seems to indicate that the problem is due to a recently made change.

**Possible implementation**
None

stdevStark, 2019-04-11:
> Just call `/recovery` (with the daemon) or `devRecoveryDevice` (with the JS library).

Maybe I solved it but for writing the tests , how to do it ? Like this is how I try it

- wallet connected
- it shows `needs seed`
- generate mnemonic
- it shows `need backup`
- recovery 

... but the value changes every time .

stdevAlDen, 2019-04-11:
> Maybe I solved it but for writing the tests , how to do it ? Like this is how I try it
>
> * wallet connected
> * it shows `needs seed`
> * generate mnemonic
> * it shows `need backup`
> * recovery
>
> ... but [...]

@stdevStark can you pls push you progress?

Senyoret1, 2019-04-11:
> Maybe I solved it but for writing the tests , how to do it ? Like this is how I try it
>
>     * wallet connected
>
>     * it shows `needs seed`
>
>     * generate mnemonic
>
>     * it shows `need [...]

The recovery procedure could be done in the third step (of after wiping the device) to recover a predefined seed. If that works, the problem I'm talking about is solved, but if you want to test a dry run with the recovery method, you can do it after recovering the predefined seed.

olemis, 2019-04-11:
@stdevAlDen @stdevAlDen start it from scratch

olemis, 2019-04-12:
> The recovery procedure could be done in the third step (of after wiping the device) to recover a predefined seed.

`setMnamonic` instead of `generateMnemonic`

olemis, 2019-04-15:
6527b84

Senyoret1, 2019-04-18:
This problem returned. I tested it with the firmware updated to 7aadfda , using the js library and the daemon.

Senyoret1, 2019-04-19:
Tested with a wrong firmware version. Closing it again.

PR #172 (2019-04-15): Fix recovery device ref #169
Fixes #169 

Changes:
- fix recovery device
- `-dryRun` required an initialized storage.

Does this change need to mentioned in CHANGELOG.md?
no

Requires changes in protobuff specs?
no

Requires testing
no
commit 6527b845 (2019-04-15): Merge pull request #172 from simelo/stdevAlDen_t169_fix_recovery_device

Fix recovery device fixes #169
closed 2019-04-19 by Senyoret1