skycoin/hardware-wallet #176: Update Firmware message, to large for Hardware Wallet Screen

stdevMac, 2019-04-13:
**Describe the bug**
When we run `go run cmd/cli/cli.go firmwareUpdate -f .../hardware-wallet/tiny-firmware/skycoin.bin`
The message resulting of a correct firmware updating is too large for the Hardware Wallet Screen

**Steps to Reproduce**
Steps to reproduce the behavior:

1. Go to the `skycoin\hardware-wallet` project
2. Run `make clean && make firmware`
3. Run `go run cmd/cli/cli.go firmwareUpdate -f path_to/skycoin/hardware-wallet/tiny-firmware/skycoin.bin` in `skycoin\harware-wallet-go` project.
4. Then follow the steps in the Hardware Wallet for a correct Updating process

**Actual behavior**
The actual message show when you run `firmwareUpdate` is 
```
"New firmware successfully installed."
"You may now unplug your Skycoin wallet."
```
From the wallet word just appear `wa`.

**Expected behavior**
The Hardware Wallet show a message according to the size of the screen.

**Possible implementation**
Change bootloader message for firmware update.
We should show a message like this:
```
"New firmware successfully installed."
"You may now unplug your SkyWallet."
```

stdevMac, 2019-04-13:
According to this issue I think we should check all the messages showed in the `Hardware Wallet`.

stdevAlDen, 2019-04-14:
@stdevMac duplicate of https://github.com/skycoin/hardware-wallet/issues/83?

stdevMac, 2019-04-14:
@stdevAlDen Yes, this is duplicated..., I already solve the problem..

stdevMac, 2019-04-14:
@stdevAlDen  Check the diff and give me some feedback

olemis, 2019-04-14:
fixed in b894b63

PR #181 (2019-04-14): Fixes #176 Update firmware message
Fixes #176 
Fixes #83 
Fixes #134 

Changes:
- Update firmware message in order to show a message according to the size of the screen.

Does this change need to mentioned in CHANGELOG.md?
yes
commit b894b638 (2019-04-14): Merge pull request #181 from simelo/stdevMac_t176_update_firmware_message

Fixes #176 Fixes #83 Fixes #134 - Skycoin wallet => Skywallet
closed 2019-04-14 by olemis