skycoin/hardware-wallet #187: Fix info for feactures response in bootloader mode

stdevAlDen, 2019-04-18:
**Describe the bug**
A clear and concise description of the problem and bug effects.

**Environment information:**

- OS: e.g. Linux, Windows, Mac OS X
- Platform: the equivalent of `uname -smr` or `uname -smrv` e.g. Darwin 17.4.0 x86_64
- Firmware Version: e.g. 0.5.18

**Steps to Reproduce**
Steps to reproduce the behavior:

1. Plug the device in bootloader mode.
2. `go run cmd/cli/cli.go features`
3. Look at the response `{"vendor":"bitcointrezor.com","major_version":1,"minor_version":5,"patch_version":0,"bootloader_mode":true,"firmware_present":true}`

**Actual behavior**
Report **bitcointrezor.com** as vendor.

**Expected behavior**
Should report skycoin as vendor.

**Possible implementation**
Just look at the code:
https://github.com/skycoin/hardware-wallet/blob/7aadfda11ba350f2e67f421d3a5845065057b322/tiny-firmware/bootloader/usb.c#L245

stdevAlDen, 2019-04-18:
> 3\. vendor

@gz-c what should be the value for `vendor` instead of `bitcointrezor.com`?

olemis, 2019-04-23:
> @gz-c what should be the value for `vendor` instead of `bitcointrezor.com`?

skycoin.net ?

stdevStark, 2019-05-03:
@stdevAlDen this is complete ?

stdevAlDen, 2019-05-03:
> @stdevAlDen this is complete ?

@stdevStark yes https://github.com/skycoin/hardware-wallet/pull/200

olemis, 2019-05-03:
fixed in 95f43f7

PR #200 (2019-05-03): Get memprotect level ref #45
Fixes #45
closes #187 

Changes:
- ask the device for the `RDP` level: `rdp_level`
- `send_msg_features` have a new field ("rdp level info") and changes bitcointrezor.com to skycoin.net
- invert logo if red level is different to `2`.

Does this change need to mentioned in CHANGELOG.md?
yes

Requires changes in protobuff specs?
yes
https://github.com/skycoin/hardware-wallet-protob/pull/49
in bootloader too, in this delta

Requires testing
yes, I need someone to help me to test this feature in a physical device but putting rdp level to 2 will **make the device unable [... cut]
closed 2019-05-03 by olemis