skycoin/hardware-wallet #188: The device is responding “PIN mismatch” when changing the PIN

Senyoret1, 2019-04-18:
**Describe the bug**
I tried to add a new PIN and changing an existing one using the `devChangePin` function of the js library and it always responded `PIN mismatch`, even when the PIN was correct and the device actually made the change.

**Environment information:**

- OS: Windows
- Platform: X64
- Firmware Version: 7aadfda

**Steps to Reproduce**
Use the `devChangePin` function of the js library for changing the PIN code and enter the same PIN when requested.

NOTE: the `devChangePin` function is currently not working as it should in the master branch, the fix is in https://github.com/skycoin/hardware-wallet-js/pull/106 . The change in that PR is just about a wrong value being passed to the hw wallet and should not be the cause for this issue.

**Actual behavior**
The function finish with the `PIN mismatch`, but the PIN is changed.

**Expected behavior**
The function should respond `PIN changed`.

**Additional context**
This was solved, as indicated in https://github.com/skycoin/hardware-wallet/issues/166#issuecomment-482161797 (I also saw it working well), but the problem returned.

**Possible implementation**
See https://github.com/skycoin/hardware-wallet/issues/166#issuecomment-482161797

stdevStark, 2019-04-19:
error in `pinmatrix_done`

```
rendering pin matrix 483926571
Decoding pin 483
Final pin value 973
rendering pin matrix 123568497
Decoding pin 123
Final pin value 123
Comparing pin 973 vs 123
```

in every case final pin had to be `123`

stdevStark, 2019-04-19:
in the code and saw other error

```
rendering pin matrix 367291485
Decoding pin 2we
Final pin value 6XX
rendering pin matrix 154269738
Decoding pin 2rt
Final pin value 5XX
Comparing pin 6XX vs 5XX
```

if two pins have all characters not in `'1' ... '9'` interval but have the same length , the pin will be all `X`s and comparing the two will be ok . Security hole .

olemis, 2019-04-19:
> if two pins have all characters not in `'1' ... '9'` interval but have the same length , the pin will be all `X`s and comparing the two will be ok . Security hole .

this is the real issue ...

olemis, 2019-04-19:
> error in `pinmatrix_done`
> 
> ```
> rendering pin matrix 483926571
> Decoding pin 483
> Final pin value 973
> rendering pin matrix 123568497
> Decoding pin 123
> Final pin value 123
> Comparing pin 973 vs 123
> ```
> 
> in every case final pin had to be `123`

This is all wrong . if the pin to set is `123` the answers to the challenges above should have been `953` and, by coincidence `123`. Other examples for the same pin value with different matrix :

| PIN matrix | Expected PIN=`123` |
|------------|---------------------|
| 514386792 | 294 |
| 638591247 | 672 |

olemis, 2019-04-19:
> I tried to add a new PIN and changing an existing one using the `devChangePin` function of the js library and it always responded `PIN mismatch`

works for me

stdevStark, 2019-04-19:
> This is all wrong .

sorry

Senyoret1, 2019-04-19:
> in the code and saw other error
>
> ```
> rendering pin matrix 367291485
> Decoding pin 2we
> Final pin value 6XX
> rendering pin matrix 154269738
> Decoding pin 2rt
> Final pin value 5XX
> Comparing pin 6XX vs 5XX [...]

That is the reason of https://github.com/skycoin/hardware-wallet/issues/151#issuecomment-481468885

stdevIvr, 2019-04-19:
I tested and it works fine for me...

olemis, 2019-04-19:
> works for me

@Senyoret1 details https://gist.github.com/olemis/a1a245e1fcd019ca6aa23bb4b34884d4

Could you please post detailed steps you followed ? Maybe this is specific to Windows ?
In any case @stdevStark detected a bug that will be fixed as related to this issue , even if not related to the details in the description .

Senyoret1, 2019-04-19:
Tested with a wrong firmware version. Closed.

PR #190 (2019-04-24): fixes #188 - Fix pin codes with bad chars and equal length pass the pin check
Fixes #188 

Changes:
- Cut pin code if invalid char found during matrix translation 

Does this change need to mentioned in CHANGELOG.md?
no

Requires changes in protobuff specs?
no

Requires testing
yes

see `test_protect.{c,h}`
commit bfb4c622 (2019-04-24): Merge pull request #190 from simelo/stdevStark_t105_pin_chars

fixes #188 - Fix pin codes with bad chars and equal length pass the pin check
closed 2019-04-19 by Senyoret1