skycoin/hardware-wallet #189: Problem when trying to erase the PIN when the device does not have one

Senyoret1, 2019-04-18:
**Describe the bug**
If you call the function for removing the current PIN (`devRemovePin` in the js library) while the device does not have a PIN, the device immediately returns `PIN removed`, without showing the confirmation screen, and then stays in a inusable state until reconnected.

**Environment information:**

- OS: Windows
- Platform: X64
- Firmware Version: 7aadfda

**Steps to Reproduce**
1) Connect a hw wallet with no PIN assigned.
2) Call the `devRemovePin` function using the JS library.
3) See the `PIN removed` response.
4) Try to call another function, like `devAddressGen` or even `devRemovePin` again.
5) The device will respond with `Unknown message read_tiny`

**Actual behavior**
After calling `devRemovePin`, calling any other function will result in a `Unknown message read_tiny` response. That is the response that is normally obtained when trying to call a function while the device is busy (normally when showing a message in the screen and waiting for user confirmation). This suggests that after calling the function the device fails to return to a usable state. Additionally, it is strange that the device indicates that the operation was performed correctly without displaying the configuration screen.

**Expected behavior**
After finishing the procedure started by `devRemovePin` it should be possible to continue using the device without having to disconnect it and reconnect it. Also, the confirmation screen should be displayed, or and error message should be returned, instead of just returning `PIN removed`.

**Additional context**

**Possible implementation**

Senyoret1, 2019-04-19:
This was tested with a wrong firmware version. After testing with a recent version, the problem of the device being unusable until reconnected is not longer pressent, but it is still responding `PIN removed` without asking for confirmation. I don't think this is a big problem, but I will keep the issue open just in case. @olemis , fell free to close it if you think it is ok to do so.

olemis, 2019-04-29:
>  it is still responding `PIN removed` without asking for confirmation.

@Senyoret1 is PIN protection set before invoking `devRemovePin` function ?

Senyoret1, 2019-04-29:
No, if the device has a PIN everything works as expected (the user is asked for confirmation).

olemis, 2019-04-30:
> No, if the device has a PIN everything works as expected (the user is asked for confirmation).

if that's the case , it's expected behavior then .

closed 2019-04-30 by olemis