skycoin/hardware-wallet #191: Prepare the firmware for automated tests with PIN operations

stdevIvr, 2019-04-19:
**Feature description**
Adding a mechanism that supports automated tests with PIN operations can be useful to detect possible bugs easily. Bugs like [this](https://github.com/skycoin/hardware-wallet/issues/188#issuecomment-484741100) can be detected before it happen.

**Possible implementation**
There are two approaches that I see for that:
1- Using the debug mode
2- Using the `EMULATOR` flag

Each case should be able to send the pin matrix to the test suite.

stdevAlDen, 2019-07-25:
This [`layoutHome`](https://github.com/skycoin/hardware-wallet/blob/92f2bb11b7a7f29773cb236c556a404ff2f72181/tiny-firmware/firmware/reset.c#L148) call may be overridden by upper layers.

stdevAlDen, 2019-07-25:
Do not send protocol responses in [this layer](https://github.com/skycoin/hardware-wallet/blob/92f2bb11b7a7f29773cb236c556a404ff2f72181/tiny-firmware/firmware/reset.c#L149).

stdevAlDen, 2019-07-25:
Remove [`msg`](https://github.com/skycoin/hardware-wallet/blob/92f2bb11b7a7f29773cb236c556a404ff2f72181/tiny-firmware/firmware/fsm_impl.c#L577) param.