skycoin/hardware-wallet #208: Test coverage for v1

olemis, 2019-05-03:
**Feature description**

test coverage and hardening needed before releasing v1

**Is your feature request related to a problem? Please describe.**
Some paths in firmware are not tested 

**Describe the solution you'd like**
More test cases and, if feasible, test coverage reports.

**Describe alternatives you've considered**
To be honest no other options considered . Open to suggestions .

**Additional context**
It all started when I detected address generation without mnemonic did not have a test case in firmware .

**Possible implementation**

- [ ] Identify untested test paths, ans add tasks
- [ ] Write test cases
- [ ] Setup compilation with test coverage
- [ ] Generate coverage reports

olemis, 2019-05-03:
@Senyoret1 @therealssj please feel free to suggest test scenarios that need to be tested before v1 .

stdevStark, 2019-05-20:
> * Unit tests for `signatures.c`

@olemis this is difficult to test

stdevStark, 2019-05-20:
> * Unit tests fir `reset.c`

`reset_init` waits for user to write data .
@olemis can we test it ?

olemis, 2019-05-20:
@stdevStark Not now without some refactorings we won't be doing in v1 , see #191 #178

stdevAlDen, 2019-05-21:
> **Feature description**
> 
> test coverage and hardening needed before releasing v1
> 
> **Is your feature request related to a problem? Please describe.**
> Some paths in firmware are not tested
> 
> **Describe the solution you'd like**
> More test cases and, if feasible, test coverage reports.
> 
> **Describe alternatives you've considered**
> To be honest no other options considered . Open to suggestions .
> 
> **Additional context**
> It all started when I detected address generation without mnemonic did not have a test case in firmware .
> 
> **Possible implementation**
> 
> * [x]  Identify untested test paths, and add tasks
> * [ ]  Write test cases
> * ~Setup compilation with test coverage~ see #212
> * ~Generate coverage reports~ see #212
> 
> Test scenarios identified
> 
> * [x]  Generate address without mnemonic set
> * [x]  Internal entropy should always be `0` beyond firmware function boundaries.
> * [x]  Entropy pool mixer should change state after processing `EntropyAck` message
> * [x]  Unit tests for `serialno.c`
> * ~Unit tests for `signatures.c`~
> * [ ]  Unit tests fir `reset.c`
> * [ ]  Unit tests for `recovery.c`

I'm having some issues running the following tests(at the right side of the arrow you can found a possible root cause):
- [ ] `SKY_cipher_PubKey_Verify` --> TestAddressVerify
- [ ] `TestPubKeyFromHex` --> TestNewPubKey and void tobuff(const char* str, uint8_t* buf, size_t bufferLength) do not make validations looking for valid characters
 - [ ] `TestNewPubKey` --> SKY_cipher_PubKey_Verify
 - [ ] `TestAddressVerify` --> unable to find secp256k1.VerifyPubkey
 - [ ] `TestPubKeyVerifyNil` --> SKY_cipher_PubKey_Verify
 - [ ] `TestPubKeyToAddress2` --> unable to find secp256k1.VerifyPubkey
 - [ ] `TestMustSigFromHex` --> void tobuff(const char* str, uint8_t* buf, size_t bufferLength) do not make validations looking for valid characters
 - [ ] `TestPubKeyFromSecKey` --> void generate_pubkey_from_seckey(const uint8_t* seckey, uint8_t* pubkey) do not check for a null seckey
 - [ ] `TestECDHonce` ???
 - [ ] `TestECDHloop` --> TestECDHonce
 - [ ] `TestSHA256KnownValue` --> a possible bug in hash function
 - [ ] `TestSHA256FromHex` --> void tobuff(const char* str, uint8_t* buf, size_t bufferLength) do not make validations looking for valid characters
 - [ ] `TestAddressBulk` --> TestAddressVerify
 - [ ] `TestNewPubKey` --> SKY_cipher_PubKey_Verify

jdknives, 2019-07-15:
@olemis This has not been touched for a while. (Or rather the PR #209 ) If Stark cannot do it for some reason, one of the Ukrainian team members could give this a shot.

olemis, 2019-07-15:
> If Stark cannot do it for some reason, one of the Ukrainian team members could give this a shot.

@stdevStark should be finishing tests for `reset.c` , he's been working on the other issues assigned to the Ukranian team .

stdevAlDen, 2019-07-16:
> > If Stark cannot do it for some reason, one of the Ukrainian team members could give this a shot.
> 
> @stdevStark should be finishing tests for `reset.c` , he's been working on the other issues assigned to the Ukranian team .
> 
> @stdevAlDen is figuring out if tests for `recovery.c` are feasible for v1 . In case there's no straightforward manner to get this done soon , we'll defer that until after v1 .

@olemis I check this as solved but actually I move it to a different issue, [here](https://github.com/skycoin/hardware-wallet/issues/315) the details, pls consider this for the next release.

commit 6e80c965 (2019-07-04): Merge pull request #209 from simelo/olemis_t208_testcov_v1

fixes #208 - Test coverage for v1

- Check that mnemonic required for address generation
- Test cases for entropy
PR #285 (2019-07-21): fixes #208 Tests for firmware before release
Fixes #208 

Changes:	
- Tests for serialno.c
- Tests for reset.c

Does this change need to mentioned in CHANGELOG.md?	
no

Requires testing	
no
commit ef839731 (2019-07-21): Merge pull request #285 from simelo/stdevStark_t208_unitests

fixes #208 Tests for firmware before release
closed 2019-07-21 by olemis